RobotLocomotion/drake #3732: Need API for getting contact points that belong to a collision group

patmarion, 2016-10-09:
This feature request is for development of the pydrake ik interface.  It is needed to initialize the quasistatic constraint with the correct subset of foot contact points.

 In the former matlab system, we had this call to get the heel and toe contacts:

    footPts = robot.body(bodyIndex).getTerrainContactPoints({'toe', 'heel'});

In c++ classes RigidBody and RigidBodyTree, it looks like a RigidBody can return all its contact points, but there isn't api to request a subset of the contact points by specifying a collision group. 

The c++ method RigidBodyTree::getTerrainContactPoints(body) is iterating over all collision elements, and it could filter by collision group at this point I suppose.

sherm1, 2016-10-09:
@SeanCurtis-TRI is working on this part of the code now -- this will be very helpful as a real use case.

avalenzu, 2016-10-09:
@SeanCurtis-TRI, Note that this is the "other type of collision group"
mentioned in the discussion of collision cliques.

SeanCurtis-TRI, 2016-10-10:
@patmarion What's the time frame for you on this?

patmarion, 2016-10-10:
Unless sometime else is interested, I am going to modify the c++ interface myself so I can get the contact points I need and continue making progress on the python side.  I don't mind if this is throwaway code, as long as there is eventually an equivalent api in place.

SeanCurtis-TRI, 2016-10-10:
I'd say go for it.  I'm currently working on that kind of functionality through the System abstraction.  You put yours in place and when I'm done on my side, I'll stitch them together.

patmarion, 2016-10-18:
I've included code in https://github.com/RobotLocomotion/drake/pull/3830 that extends the RigidBodyTree::getTerrainContactPoints() method to take an optional group name.  When merged, this issue will be resolved.

jwnimmer-tri, 2019-03-29:
Closed via #3830.

closed 2019-03-29 by jwnimmer-tri