RobotLocomotion/drake #3733: Is it (currently) possible to use the second argument of the Simulator constructor?

RussTedrake, 2016-10-09:
Attempting
```
  auto init = system->CreateDefaultContext();
  drake::systems::Simulator<double> simulator(*system,init);
```
results in
```
[100%] Building CXX object examples/CMakeFiles/simple_ct_example.dir/simple_ct_example.cc.o
/Users/russt/drake-distro/drake/examples/simple_ct_example.cc:70:55: error: call to implicitly-deleted copy constructor of 'std::unique_ptr<Context<double> >'
  drake::systems::Simulator<double> simulator(*system,init);
```

I also don't see it covered by any of the tests in the analysis/test/ directory.

sherm1, 2016-10-09:
@edrumwri, can you take a look at this? Sounds like it needs a bug fix and a test case.

david-german-tri, 2016-10-10:
I'm guessing that  `std::move(init)` in the call site will fix it.  (working-from-airport-during-nominal-vacation right now; otherwise, I'd try it myself.)  Certainly agree there should be test coverage.

edrumwri, 2016-10-12:
Yes- will handle it.

closed 2016-10-12 by edrumwri