RobotLocomotion/drake #3736: gurobi_solver does not parse multiple quadratic costs correctly

hongkai-dai, 2016-10-09:
`gurobi_solver` cannot parse multiple quadratic costs `sum_i 0.5*x'*Q_i*x+b_i'*x` on the same variable `x`. 

@naveenoid , are you also working on this? Would you like me working on it?

naveenoid, 2016-10-10:
Lets discuss f2f and then decide if you want to fix it instead.

siyuanfeng-tri, 2016-10-17:
I vaguely remember this is causing #3818, not quite sure.. I can also fix it if you guys are busy with other stuff.

hongkai-dai, 2016-10-17:
#3818 is not caused by the cost, but the constraints. The current `gurobi_solver` does not allow to impose constraints on a subset of decision variables. The multiple-costs problem is already fixed in PR #3792. I am going to close this PR.

closed 2016-10-17 by hongkai-dai