RobotLocomotion/drake #3737: Question on alternatingMIQP vs humaniods2014

samkhal, 2016-10-09:
I've been playing around with the footstep planner for Atlas, and read through the relevant paper (http://groups.csail.mit.edu/robotics-center/public_papers/Deits14a.pdf). Why do runAtlasWalking and runAtlasWalkingPlanning use alternatingMIQP, but runAtlasWalkingPlanning uses humanoids2014? 

From a cursory look it seems that alternatingMIQP uses a QP to solve for positions and then a nonlinear optimization to solve for orientation; playing around with it, though, I had trouble getting any foot orientation changes in the resulting plan. Is there any advantage to it over humanoids2014? Was one planner intended to replace the other, or are there tradeoffs?

rdeits, 2016-10-09:
We found the alternatingMIQP to be faster when dealing with unconstrained (flat ground) walking, which was the case in most of the DRC. So we used that solver as the primary footstep planner when walking with the real Atlas, and therefore it's the default for the Atlas walking tests. 

The major tradeoff is that the alternating optimizations sacrifice any notion of global optimality, especially when dealing with obstacles. The humanoids2014 solver should be more robust when in tightly constrained environments.

samkhal, 2016-10-09:
Thanks! That does clear things up. One last thing, though; as I mentioned above alternatingMIQP seems to have difficulty with turning, at least for me; for instance, running runAtlasFootstepPlanning (with alternatingMIQP) with the same navgoal as runAtlasWalkingPlanning (with a pi/2 yaw goal) results in a plan of footsteps that don't turn as you would expect. Do you know why this would be?

rdeits, 2016-10-10:
That's interesting. Can you show me exactly how you set the navgoal, and what the resulting plan was? The easiest way to draw the plan is to call `plan.draw_2d()`, as in line 33 of `runAtlasFootstepPlanning`. 

For example, I set the navgoal to pi/2 yaw with:

```matlab
goal_pos = struct('right', [1.13;0;0;0;0;pi/2], 'left', [0.87;0;0;0;0;pi/2]);
```

And got this plan: 

[image]

which is what I would expect to see. Note that I also changed the x and y positions of the navgoal for each foot, since asking for the right and left feet to end up in the same respective locations while rotating by 90 degrees would result in an awkward (in fact, unreachable) stance for Atlas. 

If you're interested in changing the behavior of the footstep planner, all of its default parameters come in from https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Atlas/%40Atlas/Atlas.m#L330 and can be changed by passing them in as `step_params` to the call to `planFootsteps()`, as follows:

```matlab
plan = r.planFootsteps(start_pos, goal_pos, [], struct('step_params', struct('max_num_steps', 10)));
```

samkhal, 2016-10-10:
My navgoal should be fine; I'm running the same parameters, but one solver turns and the other doesn't. My current install is relatively outdated, so if these planners changed since last February that might have something to do with it (last common ancestor: https://github.com/RobotLocomotion/drake/commit/26a9a6e5531e2fb46fbfb249f2e6b730f0143bad).

My code:
```
load('data/atlas_fp.mat', 'xstar');
r = Atlas('urdf/atlas_minimal_contact.urdf');
r = r.setInitialState(xstar);
kinsol = doKinematics(r, xstar(1:r.getNumPositions()));
start_pos = struct('right', forwardKin(r, kinsol, r.foot_frame_id.right, [0;0;0], 1), ...
                   'left', forwardKin(r, kinsol, r.foot_frame_id.left, [0;0;0], 1));

               
navgoal = [1;0;0;0;0;pi/2];               
R=rotz(navgoal(6));
rfoot_navgoal = navgoal;
lfoot_navgoal = navgoal;
rfoot_navgoal(1:3) = navgoal(1:3) + R*[0;-0.13;0];
lfoot_navgoal(1:3) = navgoal(1:3) + R*[0;0.13;0];

goal_pos = struct('right', rfoot_navgoal, 'left', lfoot_navgoal);
method = @footstepPlanner.humanoids2014;
%method = @footstepPlanner.alternatingMIQP;

plan = r.planFootsteps(start_pos, goal_pos, [], struct('method_handle',method));

plan.draw_2d();
```

With `method = @footstepPlanner.humanoids2014`:
[image]

With `method = @footstepPlanner.alternatingMIQP`:
[image]

samkhal, 2016-10-10:
It looks like SNOPT might be the issue; since I don't have access to the full version of snopt, I'm using the student version (7.5) as I describe here: https://github.com/RobotLocomotion/drake/issues/3369#issuecomment-245497973.

It doesn't raise any errors or warning, and doesn't fall back to fmincon with the code above. So perhaps it's to do with differences between the full and student versions of SNOPT, or between versions 7.2 and 7.5

If I remove snopt from my matlab path, the results are as follows.
alternatingMIQP falls back to fmincon, and then crashes after failing a derivative check:
```
DerivativeCheck Information

Objective function derivatives:
Maximum relative difference between user-supplied 
and finite-difference derivatives = 6.66954e-06.
  User-supplied derivative element (20,1):     -0.0127058
  Finite-difference derivative element (20,1): -0.0126991
____________________________________________________________

Error using validateFirstDerivatives (line 95)
DerivativeCheck failed:
User-supplied and forward finite-difference derivatives do not match within 1e-06 relative tolerance.
```

If I change nonlinearCollocation.m to remove the DerivativeCheck when calling fmincon, the solver finds a reasonable-looking turning plan, with the following completion message:
```
Local minimum found that satisfies the constraints.

Optimization completed because the objective function is non-decreasing in 
feasible directions, to within the default value of the function tolerance,
and constraints are satisfied to within the default value of the constraint tolerance.

<stopping criteria details>
```

So, alternatingMIQP works with fmincon if derivative checks are disabled, but fails if it's trying to use my student SNOPT 7.5. Hmm...

rdeits, 2016-10-10:
Okay, interesting. Thank you for figuring that out! I don't think we'll have cycles to fix the student snopt issue, since the long-term plan is to move the footstep planner out of Matlab entirely and into the new Drake C++ optimization framework. But I'm glad to know that fmincon is able to at least give a reasonable answer with the derivative checks turned off. It sounds like turning off those derivative checks for fmincon would be a good idea, although I'm not sure how to detect or fix the problem with student snopt just silently being wrong.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri