RobotLocomotion/drake #3739: Broken CI - linux-clang-nightly-coverage - Undefined LLVM references When Running avl Unit Test

liangfok, 2016-10-10:
# Problem Definition

[linux-clang-nightly-coverage](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-nightly-coverage/) has been broken since October 6, 2016. The error relates to LLVM undefined symbols when trying to run one of `avl`'s unit tests:

```
03:38:16 "/usr/local/bin/ctest" --launch --target-name zoomtest --build-dir /home/ubuntu/workspace/linux-clang-nightly-coverage/build/externals/avl/plotlib/examples --output zoomtest --target-type EXECUTABLE --language Fortran -- /usr/bin/gfortran-4.9   -fprofile-arcs -ftest-coverage  -std=legacy -g CMakeFiles/zoomtest.dir/zoomtest.f.o  -o zoomtest  -L/us../libplt.a(Xwin.c.o): In function `__llvm_gcov_writeout':
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31c2): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31d6): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x31f7): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x320b): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x322c): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3240): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3261): undefined reference t /home/ubuntu/workspace/linux-clang-nightly-coverage/build/externals/avl/plotlib/examples && "/usr/local/bin/ctest" --launch --target-name symbols --build-dir /home/ubuntu/workspace/linux-clang-nightly-coverage/build/externals/avl/plotlib/examples --output CMakeFiles/symbols.dir/symbols.f.o --source /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/examples/symbols.f --language Fortran -- /usr/bin/gfortran-4.9    -std=legacy -g   -c /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/examples/symbols.f -o CMakeFiles/symbols.dir/symbols.f.o
03:39:09 [ 19%] Linking Fortran executable symbols
03:39:09 cd /home/ubuntu/workspace/linux-clang-nightly-coverage/build/externals/avl/plotlib/examples && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/symbols.dir/link.txt --verbose=1
03:39:09 "/usr/local/bin/ctest" --launch --target-name symbols --build-dir /home/ubuntu/workspace/linux-clang-nightly-coverage/build/externals/avl/plotlib/examples --output symbols --target-type EXECUTABLE --language Fo `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3275): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3296): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x32aa): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x32cb): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x32df): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3300): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3314): undefined reference to `llvm_gcda_eortran -- /usr/bin/gfortran-4.9   -fprofile-arcs -ftest-coverage  -std=legacy -g CMakeFiles/symbols.dir/symbols.f.o  -o symbols  -L/usr/local/lib ../libplt.a /usr/lib/x86_64-linux-gnu/libSM.so /usr/lib/x86_64-linux-gnu/libICE.so /usr/lib/x86_64-linux-gnu/libX11.so /usr/lib/x86_64-linux-gnu/libXext.so 
03:39:09 mit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3335): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3349): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x336a): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x337e): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x339f): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x33b3): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x33d4): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x33e8): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3409): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x341d): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x343e): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3452): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3473): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3487): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34a8): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34bc): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34dd): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x34f1): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3512): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3526): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3547): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x355b): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x357c): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x3590): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35b1): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35c5): undefined reference to `llvm_gcda_emit_arcs'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35e6): undefined reference to `llvm_gcda_emit_function'
03:39:09 /home/ubuntu/workspace/linux-clang-nightly-coverage/externals/avl/plotlib/Xwin.c:(.text+0x35fa): undefined reference to `llvm_gcda_emit_arcs'
[... cut ...] -o zoomtest  -L/usr/local/lib ../libplt.a /usr/lib/x86_64-linux-gnu/libSM.so /usr/lib/x86_64-linux-gnu/libICE.so /usr/lib/x86_64-linux-gnu/libX11.so /usr/lib/x86_64-linux-gnu/libXext.so 
```

The broken build has the following additional flags: `-fprofile-arcs -ftest-coverage`. It is also missing the following linker flags: `-L/usr/local/lib ../libplt.a /usr/lib/x86_64-linux-gnu/libSM.so /usr/lib/x86_64-linux-gnu/libICE.so /usr/lib/x86_64-linux-gnu/libX11.so /usr/lib/x86_64-linux-gnu/libXext.so`

jamiesnape, 2016-10-10:
I just disabled them on Clang coverage builds. Probably some strange issue because the Fortran code is compiled with GCC and everything else Clang. We still get coverage on GCC builds.

liangfok, 2016-10-10:
Thanks! I'll close this issue now.

closed 2016-10-10 by liangfok