RobotLocomotion/drake #3743: System 2.0 PID Controller Needs to Support a Vector of Gains

liangfok, 2016-10-10:
System 2.0's PID controller currently assumes the same gain can be applied to every control signal. This will not work when the control signals are applied to very different systems like the steering angle and throttle speed of a car that require different gains.

Addressing this issue will require generalizing `gain.h`, and `pid_controller.h`, among others.

sherm1, 2016-10-10:
@liangfok, you can have as many PID controllers as you want in a Diagram and each has its own gains. Why isn't that a good enough solution?

liangfok, 2016-10-10:
Having multiple PID systems will require that I have mux / demux systems to "break apart" and "re-combine" signals that would otherwise be a vector (I think). For example, suppose I have three actuators in a `RigidBodyTree`:

1. front left wheel
2. front right wheel
3. steering

The states of these are represented by a single vector in `RigidBodyPlant`. I will need to separate these signals into separate PID controllers to assign them different gains. I figured it would be cleaner to support a vector of gains within a PID controller.

sherm1, 2016-10-11:
It would be great to see these parameters delivered on input or parameter ports like [Simulink's PID controller](https://www.mathworks.com/help/simulink/slref/pidcontroller.html). I'm not as excited about feeding these into the constructor.

david-german-tri, 2016-10-11:
Feeding them into the constructor seems like a fine interim solution to me.  Parameters are coming relatively soon, but I wouldn't want to block this on them.

liangfok, 2016-10-12:
I plan to work on this issue after #3742 is merged.

liangfok, 2016-10-14:
Note that once #3619 is resolved, the gains can be extracted out of `drake::systems::PidController` thus resolving this issue.

liangfok, 2016-10-14:
Thinking about this some more, I think [System 2.0's PID controller](https://github.com/RobotLocomotion/drake/blob/16272e5953eaf4068747b5242d3eef9ee1fa7191/drake/systems/controllers/pid_controller.h) can be greatly simplified by keeping the gains (as vectors) within it and making it a `System` instead of a `Diagram`. The equations are simple enough that they can be directly implemented within a single system. Note that [System 1.0's PID controller](https://github.com/RobotLocomotion/drake/blob/d9b896925e58661c49b480e607ab8a528ceb3fb7/drake/systems/pd_control_system.h) is a single system and [uses vectors for gains](https://github.com/RobotLocomotion/drake/blob/d9b896925e58661c49b480e607ab8a528ceb3fb7/drake/systems/pd_control_system.h#L85).

RussTedrake, 2016-10-14:
sounds like a good idea to me.

amcastro-tri, 2016-10-21:
I agree with you @liangfok. The reason I originally implemented the PID controller this way was during the sprint to test we could build complicated Diagram systems composed of Diagram's (the concept of Diagram being a tree of Diagram's). This PID controller was the simplest useful system example I came up with. That then led me to write the more complex controlled Kuka using @naveenoid's gravity commentator and your own bot visualizer.

PR #3846 (2016-10-24): Generalized gain system to support a vector of gains containing different values.
I also generalized the PID controller to make use of the generalized gain system.

Closes #3743.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3846)
closed 2016-10-24 by liangfok