RobotLocomotion/drake #3749: Tried to install drake for the first time, addpath_drake fails

Tilstandsrommet, 2016-10-10:
`addpath_drake` produces the following:

> Error using addpath_drake (line 17)
> The Drake build outputs are not on the MATLAB path, and could not be auto-detected. Please add the build output directory "matlab" to the path, then re-run addpath_drake.

I tried to add "matlab" to MATLAB search path, but it did not make a difference.
[image]

RussTedrake, 2016-10-10:
the build outputs path it is referring to is

drake-distro/build/install/matlab

Tilstandsrommet, 2016-10-10:
Thanks. It works!

closed 2016-10-10 by Tilstandsrommet

SwapZap, 2017-02-23:
I am not able to find drake-distro/build/install/matlab.
Please help me, I am unfamiliar with installing software using make.

hongkai-dai, 2017-02-23:
@SwapZap we have dropped support for Windows, drake works on Ubuntu and OSX

hongkai-dai, 2017-03-21:
@orsoromeo , could you try using bazel to build your drake? You may follow the instruction on http://drake.mit.edu/bazel.html

hongkai-dai, 2017-03-21:
@orsoromeo , sorry bazel does not install MATLAB version of drake. I think you can use cmake to build drake, by following the instruction in http://drake.mit.edu/from_source.html#build-with-make

orsoromeo, 2017-03-22:
@hongkai-dai It works by building with cmake, thanks a lot! 
Does this mean that the MATLAB version of drake cannot be built as a ros catkin package?

hongkai-dai, 2017-03-23:
@orsoromeo I think nobody ever tested it to build as a ros catkin package.

If you are using MATLAB version of drake, while ROS is not written in MATLAB, do you need to build the MATLAB version of drake as a ros catkin package?

orsoromeo, 2017-03-24:
@hongkai-dai I believe that no, if I use the Matlab version of Drake then I don't need to build with ROS catkin package.
I attempted to do it because I would like to use the Matlab version of Drake and then send through ROS messages the results to other subscribers written in other languages. Is it possible to do it if I build with make?