RobotLocomotion/drake #3752: Mysterious CI Failure: RPC failed; curl 56 SSLRead() return error -36

liangfok, 2016-10-10:
# The Error

```
14:01:06 Cloning into 'externals/bullet'...
14:01:06 error: RPC failed; curl 56 SSLRead() return error -36
14:01:06 fatal: The remote end hung up unexpectedly
14:01:06 fatal: early EOF
14:01:06 fatal: index-pack failed
14:01:06 fatal: clone of 'https://github.com/RobotLocomotion/bullet3.git' into submodule path 'externals/bullet' failed
```

# Example Build Log

* https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-ninja-continuous-release/649/

# Prognosis

Network hiccup?

liangfok, 2016-10-11:
Since I'm fairly sure this is due to a transient network failure, I'm going to close this issue for now.

closed 2016-10-11 by liangfok

mwoehlke-kitware, 2016-10-11:
Happened again, also OS X:
https://drake-jenkins.csail.mit.edu/job/mac-clang-ninja-continuous-release/657

mwoehlke-kitware, 2016-10-12:
https://drake-jenkins.csail.mit.edu/job/mac-clang-ninja-continuous-debug/674
https://drake-jenkins.csail.mit.edu/job/mac-clang-ninja-continuous-release/666

...does seem to be an OS X (MacStadium?) issues...

liangfok, 2016-10-12:
How about let's modify the submodule URLs for public repositories to use `http://` instead of `https://`? Wouldn't that eliminate the `SSLRead()` error?

jamiesnape, 2016-10-12:
```
> GET /RobotLocomotion/drake.git HTTP/1.1
> Host: github.com
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
< Content-length: 0
< Location: https://github.com/RobotLocomotion/drake.git
```

liangfok, 2016-10-12:
Bummer.

liangfok, 2016-10-12:
Would switching to SSH access make a difference?

jwnimmer-tri, 2016-10-12:
The only reasonable solutions are (1) more automatic retries, or (2) find a hosting solution that sucks less.

liangfok, 2016-10-12:
Maybe try `get-retry`?
https://commondatastorage.googleapis.com/chrome-infra-docs/flat/depot_tools/docs/html/git-retry.html

Lucy-tri, 2016-12-05:
Here is another example:
https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=235736

Lucy-tri, 2016-12-05:
https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=236267
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-ninja-continuous-debug/993/

liangfok, 2016-12-12:
Again: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-ninja-nightly-matlab-release/206/

soonho-tri, 2016-12-14:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-ninja-continuous-debug/1055

liangfok, 2016-12-14:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-ninja-continuous-debug/1055/

soonho-tri, 2016-12-14:
@liangfok :-)

liangfok, 2016-12-14:
:+1: :smile:

liangfok, 2017-01-02:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-ninja-continuous-release/1150/

liangfok, 2017-01-27:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-ninja-continuous-release/1285/

liangfok, 2017-02-14:
https://drake-jenkins.csail.mit.edu/job/mac-clang-ninja-continuous-release/1394/console

liangfok, 2017-03-16:
https://drake-jenkins.csail.mit.edu/job/mac-clang-ninja-continuous-release/1585

liangfok, 2017-03-27:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-experimental-debug/2140

liangfok, 2017-04-05:
Similar error here: https://drake-jenkins.csail.mit.edu/job/linux-clang-experimental-memcheck-lsan/1209/

```
11:12:14 "/usr/local/bin/ctest" --launch --target-name convert_to_obj --build-dir /home/ubuntu/workspace/linux-clang-experimental-memcheck-lsan/build/externals/meshconverters --output CMakeFiles/convert_to_obj.dir/vcglib/wrap/ply/plylib.cpp.o --source /home/ubuntu/workspace/linux-clang-experimental-memcheck-lsan/externals/meshconverters/vcglib/wrap/ply/plylib.cpp --language CXX -- /usr/bin/clang++  -DQT_CORE_LIB -DQT_XMLPATTERNS_LIB -DQT_XML_LIB -I/home/ubuntu/workspace/linux-clang-experimental-memcheck-lsan/externals/meshconverters/vcglib -isystem /usr/include/qt4/QtCore -isystem /usr/share/qt4/mkspecs/default -isystem /usr/include/qt4 -isystem /usr/include/qt4/QtXml -isystem /usr/include/qt4/QtXmlPatterns  -fsanitize=leak -fsanitize-blacklist=/home/ubuntu/workspace/linux-clang-experimental-memcheck-lsan/tools/blaerror: RPC failed; result=56, HTTP code = 200
11:12:14 fatal: The remote end hung up unexpectedly
11:12:14 fatal: early EOF
11:12:14 fatal: index-pack failed
11:12:14 Clone of 'https://github.com/RobotLocomotion/bullet3.git' into submodule path 'externals/bullet' failed
11:12:14 error: RPC failed; result=56, HTTP code = 200
11:12:14 fatal: The remote end hung up unexpectedly
11:12:14 fatal: early EOF
11:12:14 fatal: index-pack failed
11:12:14 error: RPC failed; result=56, HTTP code = 200
11:12:14 Clone of 'https://github.com/OctoMap/octomap.git' into submodule path 'externals/octomap' failed
11:12:14 fatal: The remote end hung up unexpectedly
```