RobotLocomotion/drake #3766: Implement by-name accessors to rigid-body state members

ggould-tri, 2016-10-11:
In #3702 @siyuanfeng-tri recently wrote the following code:
```
   name_to_velocity_index_ = std::unordered_map<std::string, int>();
    for (int i = 0; i < robot_.get_num_velocities(); i++) {
      name_to_velocity_index_[robot_.get_velocity_name(i)] = i;
    }
    for (int i = 0; i < static_cast<int>(robot_.actuators.size()); i++) {
      actuator_name_to_actuator_index_[robot_.actuators.at(i).name_] = i;
    }
```
...
```
   nominal_position_[name_to_position_index().at("rightHipRoll")] = 0.01;
```
I recently wrote similar code in my Schunk simulator branch.

This seems like a common use case and should be part of the RBS/RBT/RBP APIs.

liangfok, 2016-10-11:
One problem with a simple string is there may be multiple instances of the same model in a `RigidBodyTree`. To resolve this, either the string needs to be a 2-tuple or the string needs to include the unique model instance ID.

ggould-tri, 2016-10-11:
... or the method needs to return a list of matching elements, or it needs to return the one of that name if unique and null otherwise, or multiple instances need to have distinct name prefixes set at load-time.  There are any number of ways to solve that.

Given that the common case is a single load of a single SDF/URDF with internally unique names, we should have some accessor which makes that case easy.

liangfok, 2016-10-11:
How about adding the following?

```
typedef std::map<std::string, int> IndexTable;
IndexTable RigidBodyTree::GetPositionNameToIndexTable(int model_instance_id);
IndexTable RigidBodyTree::GetVelocityNameToIndexTable(int model_instance_id);
IndexTable RigidBodyTree::GetActuatorNameToIndexTable(int model_instance_id);
```

ggould-tri, 2016-10-11:
From a user point of view, the user just has a string that they entered in the sdf and probably wants to be able to index with that.  For a single-model SDF, here's how that looks in your proposal:
```
RigidBodyTree tree;
int model_id = AddModelInstancesFromSdfFileInWorldFrame(... &tree).begin()->second;
[...do some things...]
auto widget_velocity = state[tree.GetVelocityNameToIndexTable(model_id)["widget"]];
auto gadget_velocity = state[tree.GetVelocityNameToIndexTable(model_id)["gadget"]];
auto gizmo_velocity = state[tree.GetVelocityNameToIndexTable(model_id)["gizmo"]];
...
```
I think that's pretty good.  The `GetWithReallyLongName` methods should return constref so we don't construct/destruct for every single lookup.

Is there a reason not to further factor the deference and just have `tree.velocity_index(model_id, "widget")` instead of `tree.GetVelocityNameToIndexTable(model_id)["widget"]`?  It looks clearer on the screen IMO.

liangfok, 2016-10-11:
I suppose it would be OK to modify `RigidBodyTree` to own the various index tables, which are then accessed via the following methods:

```
int RigidBodyTree::actuator_index(int model_instance_id, std::string element_name);
int RigidBodyTree::position_index(int model_instance_id, std::string element_name);
int RigidBodyTree::velocity_index(int model_instance_id, std::string element_name);
```

The reason I didn't consider this before is because the `ModelInstanceId` table is currently not stored in the `RigidBodyTree`. It's owned by the application that uses the tree.

I'd like get @sherm1's opinion on this since it involves adding methods to `RigidBodyTree`.

sherm1, 2016-10-11:
@ggould-tri, @liangfok: I prefer that we not attempt to add this functionality to RBTree, but rather encapsulate it in a separate class that can be customized for the use case you have in mind. RBTree has to be very general and deal with multi-dof joints, joints whose coordinates don't make sense individually (because of quaternions, for example) and lots of other things that haven't been implemented yet. We are planning in Q4 on ripping up RBTree into several pieces, changing its API substantially, and generally mucking around with it. A separate name-to-info class could be useful right now and continue to be useful later, regardless of simplifying assumptions it might make.

liangfok, 2016-10-12:
@sherm1, thanks for the advice! I see your point about how things can get complicated with multi-DOF joints and quaternion representations.

@ggould-tri, I now think the application will have to maintain ownership of the various index tables. We can provide helper methods that create the tables. These helper methods can make use of `RigidBodyTree`'s existing API.

ggould-tri, 2016-10-12:
I like @sherm1 's suggestion: if RBT can be a purely algorithmic entity devoid of frills, and can be interacted with from outside of the System via a wrapper class with things like naming and unique index semantics.  If we're building a separate helper class with the convenience accessors, that class can also own all of the lookup tables.

liangfok, 2016-10-12:
Suggested name of wrapper: `RigidBodyTreeWithModelsContainingOnlyOneDOFJoints`

ggould-tri, 2016-10-12:
...aaaand we're done here.

Closing this for now since I don't think this bug corresponds to concrete upcoming work.  At least until I start seeing rule-of-three violations in my PR reviews.

closed 2016-10-12 by ggould-tri