RobotLocomotion/drake #3773: support an open-source SDP solver in c++

RussTedrake, 2016-10-12:
One candidate is https://github.com/cvxgrp/scs .

sherm1, 2016-10-12:
That looks great! MIT licensed too.

jamiesnape, 2016-10-12:
SCS sounds good. We would probably need to try to work with them on their build system.

RussTedrake, 2016-10-12:
this is a longer term goal.  we will get the SDP tools working in c++ with gurobi/mosek first.

rdeits, 2016-10-12:
I've used SCS for some basic sums-of-squares programming, and it's worked well. I'm using it now to run the unit tests in https://github.com/rdeits/Mayday.jl

RussTedrake, 2016-10-13:
great list of solver accessible in julia is here:
http://www.juliaopt.org/#solvers

fwiw - @rdeits says this is the best eco-system of solvers that he has seen yet, and it is quite similar to what we’ve been building in MathematicalProgram.

hongkai-dai, 2016-12-02:
@RussTedrake , do you need C++ Drake for your class next semester? If so, we will probably need to bring SCS into Drake, as a non-commercial solver for the students.

hongkai-dai, 2017-09-08:
In order to bring SCS https://github.com/cvxgrp/scs into drake, I can see there are two things we need to do. 

1. Bazel or cmake needs to checkout out and build SCS.
2. We need to write our wrapper 'SCSSolver` to pass problem information to SCS.

@jamiesnape @jwnimmer-tri could you help me with the first one, to build SCS using Bazel and cmake?

jamiesnape, 2017-09-08:
At this stage we can skip the CMake, but sure, I can help with the Bazel.

hongkai-dai, 2017-11-06:
@jamiesnape , I wish we could add SCS as a solver to Drake. Is it OK if you could help with the Bazel build of SCS?

jamiesnape, 2017-11-07:
Sure, direct or indirect solver or both? For the record, looks like the direct solver contains LGPL v2.1 code.

jamiesnape, 2017-11-07:
From Slack conversation, we are going to start with the indirect solver and look at making both available eventually.

hongkai-dai, 2017-11-07:
The SCS paper https://web.stanford.edu/~boyd/papers/pdf/scs.pdf mentions that the direct method is more accurate than the indirect method, as the direct method uses an LDL decomposition to solve the linear system `Ax = b` directly and exactly, whereas the indirect method solves this linear system through conjugate gradient method indirectly and approximately. The computation time of the two approaches are comparable, as shown in the examples in the paper. The difference in the computation time between the direct and indirect methods, are mainly caused by the sparsity of the linear system `Ax = b`. When the linear constraint in our problem has a nice sparsity pattern, i.e., the number of nonzeros are similar across different columns of `A`, we should expect the indirect method to be faster than the direct method.

hongkai-dai, 2018-11-04:
Currently the open source SDP solver (in C/C++) include CSDP, DSDP and SDPA. This [website](http://plato.asu.edu/bench.html) offers some good comparison on benchmark problems.

According to http://plato.asu.edu/ftp/sparse_sdp.html, the average performance is CSDP > SDPA > DSDP.

tri-ltyyu, 2019-03-26:
We currently have SAS which is low precision. We'd still like a high precision solver, so leaving this in the backlog.

jamiesnape, 2019-07-05:
@hongkai-dai Was this closed by https://github.com/RobotLocomotion/drake/pull/11711?

hongkai-dai, 2019-07-05:
Close the issue as #11711 resolved it.

closed 2019-07-05 by hongkai-dai