RobotLocomotion/drake #3777: Mysterious CI Failure - Undefined function 'parseparams' for input arguments of type 'cell'.

liangfok, 2016-10-12:
# The Problem

Example 1:

```
Undefined function 'parseparams' for input arguments of type 'cell'.

Error in mesh (line 53)
[reg, prop]=parseparams(args);

Error in RigidBodyHeightMapTerrain/plotTerrain (line 66)
      mesh(reshape(xyz(1,:),size(X)),reshape(xyz(2,:),size(X)),reshape(xyz(3,:),size(X)));

Error in terrainInterpTest (line 20)
  plotTerrain(options.terrain);

CMake Error at /usr/local/share/cmake-3.7/Modules/MatlabTestsRedirect.cmake:104 (message):
  [MATLAB] TEST FAILED Matlab returned 1
```

Example 2:

```
Error using quiver (line 44)
Undefined function 'parseparams' for input arguments of type 'cell'.

Error in testFootstepSolvers/test_solver (line 74)
  quiver(h, steps(1,r_ndx), steps(2, r_ndx), cos(steps(6,r_ndx)), sin(steps(6,r_ndx)), 'b', 'AutoScaleFactor', 0.2)

Error in testFootstepSolvers (line 92)
test_solver(@footstepPlanner.footstepMIQP, h, 'miqp');

CMake Error at /usr/local/share/cmake-3.7/Modules/MatlabTestsRedirect.cmake:104 (message):
  [MATLAB] TEST FAILED Matlab returned 1
```

Example 3:

```
Error using quiver (line 44)
Undefined function 'parseparams' for input arguments of type 'cell'.

Error in testFootstepSolversNoRegions/test_solver (line 57)
  quiver(h, steps(1,r_ndx), steps(2, r_ndx), cos(steps(6,r_ndx)), sin(steps(6,r_ndx)), 'b', 'AutoScaleFactor', 0.2)

Error in testFootstepSolversNoRegions (line 69)
test_solver(@footstepPlanner.footstepMIQP, h, 'miqp');

CMake Error at /usr/local/share/cmake-3.7/Modules/MatlabTestsRedirect.cmake:104 (message):
  [MATLAB] TEST FAILED Matlab returned 1
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-gcc-ninja-nightly-matlab-debug/70/

# Prognosis

MATLAB toolbox license issue?

jamiesnape, 2016-10-12:
MATLAB license server problem.

closed 2016-10-12 by jamiesnape

david-german-tri, 2016-10-12:
@liangfok You're welcome to scribe these to an issue if you feel so moved, but you should also feel free not to, in the cases where you recognize an unresolvable root cause (e.g. license server flakiness).

liangfok, 2016-10-12:
OK. I'm recording these errors as issues so we can maintain a history to get a sense of whether the problem is worth fixing.

jamiesnape, 2016-10-12:
Maybe consolidate them into a single issue, but it is not that it is not worth fixing, more that it is basically impossible to fix. A license server is the only option for MATLAB. We tried every single other option, there, and some network disruption is inevitable in the cloud. MathWorks or FlexNet would have to solve the problem of dealing better with that.

liangfok, 2016-10-12:
I see. OK. I'll consolidate these into a single issue. Is the same license server being used for all of the various licenses? In the errors linked to above, MATLAB starts, meaning it got a license, but some other license failed. I'm having difficulty discerning which license acquisition failed.

Also, is it possible to modify these unit tests to declare which licenses it requires so the unit tests can fail fast when they fail to acquire all required licenses?