RobotLocomotion/drake #3778: Unable to Tell Which Optimizer Was Used → Difficult to Debug Problems

liangfok, 2016-10-12:
# Problem Definition

I often encounter a problem where a particular unit test works in one configuration but fails in another. See, for example:

* Works: https://drake-cdash.csail.mit.edu/testDetails.php?test=3517551&build=145349
* Fails: https://drake-cdash.csail.mit.edu/testDetails.php?test=3516356&build=145327

From looking at the output, I can't tell _why_ the failure occurred. Given the configuration difference, I'm fairly certain the failure is due to the selection of a different (open-source) optimizer, but I can't be certain because the output doesn't indicate which optimizer was used.

# Suggested Resolution

Add a print statement indicating which optimizer was selected before performing the optimization.

liangfok, 2016-10-12:
FTR, I'm pretty sure the above failure is tracked by #3711.

liangfok, 2016-10-12:
@ggould-tri I'm going to assign this issue to you because I believe you're familiar with the optimizer selection process.

sammy-tri, 2016-10-12:
I believe MathematicalProgram::GetSolverResult() will return the name of the solver which was used.  This isn't going to percolate all of the way through MATLAB up to the CDash output, but it may be a start...

ggould-tri, 2016-10-12:
Logging the optimizer for each solve is impractical, as there are use cases that will do solves at the kHz level.  I will think about how best to solve this.  It might be that I could log it at low priority if CI tests set the log verbosity, but I'm not sure if that is practical or wise.

ggould-tri, 2016-10-12:
Scribing @jwnimmer-tri notes on how to turn on verbose logging for unit tests, which is probably on the right track to the best way to fix this issue:

> You can replace GTest::Main with your own hand-written main (which could share across Drake), that both parses the GTest flags, as well as GFlags flags (for which libcommon provides a logging toggle).
> 
> GTest and GFlags fight.  Ask stackoverflow for which order to feed them the flags.
> 
> In CMake, `GTest::Main` is a library that provides `int main()`.  Instead, we write our own "library" that provides `int main()`.

RussTedrake, 2016-10-12:
to be clear, the solvers in those examples are in matlab (they do not touch mathematicalprogram in any way).  the failure is in the PATH solver, but path is probably not the problem... it just means that the robot fell down.  the difference in configuration looks like the first case had gurobi and snopt and the second did not... so probably the initial conditions solution was different in the two cases... which ultimately caused the robot to fall down (making PATH fail).  

the other possibility is that during execution, the fastqp tried to fall back to the QP solver.  since gurobi was not present, it used a different solver.  which gave a different result.  and the robot eventually fell down.

this is far beyond what you should try to debug.  i would see if guarding the test with WITH_SNOPT and/or WITH_GUROBI.

ggould-tri, 2016-10-13:
Ah, I assumed that since I was assigned that someone had determined it was a C++-side issue.  Since it is matlab, I am going to unassign myself and tag it to RLG.

(We probably ought to do the described C++ work sometime but, as @jwnimmer-tri says, it "
is several yaks deep," so _entia non sunt multiplicanda praeter necessitatem_.)

liangfok, 2016-10-13:
Sorry I thought MATLAB was calling C++'s `MathematicalProgram`. Is that the eventual plan?

sammy-tri, 2016-10-13:
I guess I should have taken a closer look at the backtrace here, I think I just assumed it was in some other place.  I don't think there's a plan to port TimeSteppingRigidBodyManipulator to MathematicalProgram...   At least not any time soon.

RussTedrake, 2016-10-13:
That's not quite right -- we have lcp in mathematicalprogram.  But it's probably a longer answer than belongs on this issue.

hongkai-dai, 2016-10-18:
This solver issue is also discussed in #3817

sammy-tri, 2016-12-22:
Closing, I'm not planning on fixing MATLAB debugging issues.

closed 2016-12-22 by sammy-tri