RobotLocomotion/drake #3813: Legacy support, drakeSnopt.zip

mposa, 2016-10-15:
I have an older branch, which I'm trying to install. Unless I've screwed something up, it's attempting to download SNOPT at the following location (and failing):

http://drake002.csail.mit.edu/drake/drakeSnopt.zip

Is it possible that this file was just pulled from the server? Thanks!

liangfok, 2016-10-16:
Yes, it does appear as though the file was removed from the `drake002.csail.mit.edu` server.

Check out this PR #3767, which removed official support of pre-compiled SNOPT from the Drake. It lists the following URL, which still works:

https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip

Maybe you can modify your older branch to point to the above URL?

mposa, 2016-10-16:
Thanks. Problem solved!

closed 2016-10-16 by mposa