RobotLocomotion/drake #3814: Ubuntu installation

MoyinOtubela, 2016-10-16:
I'm having an issue regarding using make as instructed in http://drake.mit.edu/from_source.html and http://drake.mit.edu/ubuntu_trusty.html

I've also set up my environment variables in my ~/.bashrc file as such:

export CC=gcc-4.9 CXX=g++-4.9 FC=gfortran-4.9 F77=gfortran-4.9

The problem is experienced when I try to compile in my build folder in the drake-distro directory, of which I got from "git clone git@github.com:RobotLocomotion/drake.git drake-distro", after using "cmake .." and  "make". Upon using make, it seems to fails when it attempts to link a certain shared module. This is as illustrated in the figure below:

[image]

These are other details that relate to my computer environment:
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

MATLAB 9.1.0.441655 (R2016b)

java version "1.8.0_101"
Java(TM) SE Runtime Environment (build 1.8.0_101-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.101-b13, mixed mode)

I wait eagerly for a response. Thank you for your time.

liangfok, 2016-10-16:
I don't know what the problem is yet but on my computer `OcTreeNode::createChild()` also does not exist:

```
$ cd drake-distro-1/externals/octomap/octomap/lib
$ $ nm -gC liboctomap.so.1.8.0 | grep OcTreeNode::
000000000003e0a0 T octomap::OcTreeNode::addValue(float const&)
000000000003deb0 T octomap::OcTreeNode::OcTreeNode()
000000000003deb0 T octomap::OcTreeNode::OcTreeNode()
000000000003dec0 T octomap::OcTreeNode::~OcTreeNode()
000000000003dec0 T octomap::OcTreeNode::~OcTreeNode()
0000000000050850 T octomap::ColorOcTreeNode::updateColorChildren()
00000000000504d0 T octomap::ColorOcTreeNode::readData(std::istream&)
0000000000020190 T octomap::CountingOcTreeNode::CountingOcTreeNode()
0000000000020190 T octomap::CountingOcTreeNode::CountingOcTreeNode()
00000000000201a0 T octomap::CountingOcTreeNode::~CountingOcTreeNode()
00000000000201a0 T octomap::CountingOcTreeNode::~CountingOcTreeNode()
0000000000050940 T octomap::operator<<(std::ostream&, octomap::ColorOcTreeNode::Color const&)
000000000003dfa0 T octomap::OcTreeNode::getMaxChildLogOdds() const
000000000003ded0 T octomap::OcTreeNode::getMeanChildLogOdds() const
0000000000050510 T octomap::ColorOcTreeNode::getAverageChildColor() const
0000000000050490 T octomap::ColorOcTreeNode::writeData(std::ostream&) const
```

liangfok, 2016-10-16:
Is ROS being sourced (e.g., `source /opt/ros/indigo/setup.bash`)? I just heard that sourcing ROS may cause this problem.

liangfok, 2016-10-16:
I'd like to know which `octomap` library you're linking against, please modify the following file:

https://github.com/RobotLocomotion/drake/blob/master/drake/systems/robotInterfaces/CMakeLists.txt#L25

Immediately after the line containing `find_package(octomap)`, can you add the following debug print statements?

```
message(STATUS "octomap_LIBRAY=${${_variableName}}")

```

Let me know what they output.

liangfok, 2016-10-16:
With @hongkai-dai and @siyuanfeng-tri's help, I'm pretty sure the problem is due to a conflict between ROS Indigo's version of `octomap` (1.6.9) versus Drake's version (https://github.com/OctoMap/octomap/pull/140/commits/21d159f3c0bbdc3d2a3283ae4c0fd719f38bdc57). Somehow, it looks like `octomapWrapper.cpp` is being compiled using ROS's version but linked against Drake's version resulting in the undefined symbol error. I believe a fix would be to modify Drake's build system to prioritize the version in Drake's super-build over the one in ROS.

@jamiesnape and @mwoehlke-kitware any ideas?

liangfok, 2016-10-16:
FTR, the temporary workaround is to disable ROS Indigo. This can be done as follows:

First ensure your `~/.bashrc` does not contain a line with `source /opt/ros/indigo/setup.bash` (or the line is commented out). Then close your `bash` terminal and open a new one.

liangfok, 2016-10-16:
If the problem is indeed due to a conflict with ROS Indigo, perhaps `ros-indigo-octomap` needs to be installed on Drake's [linux-gcc-experimental-matlab-ros](https://drake-jenkins.csail.mit.edu/view/Experimental/job/linux-gcc-experimental-matlab-ros/) CI image so this problem can be detected.

MoyinOtubela, 2016-10-16:
Excellent. The compiling issue was resolved quickly following the advice. As suggested I edited my ~/.bashrc file where I commented out "source /opt/ros/indigo/setup.bash". I also took time to edit the CMakeLists.txt file to insert the debugging method as illustrated in the figure below. But since it's resolved I don't see the need for it at this point. It even installed successfully when installing with gurobi set on.

[image]

 "cmake .. -DWITH_GUROBI=ON && make " 

Here's the output as a result:

[image]

For the request pertaining to the finding out which outmap library that I was linking to.. This was the output I found after much scrolling:

[image]

I'm not sure about this output in general and what it means further down the line when I use drake for my research.

Thanks!

- [ ]

liangfok, 2016-10-16:
Great! Thanks for the results and output. I'm unsure why the `octomap_Foo` print statements indicate nothing is being found. Maybe I wasn't using the right variable names? One thing you can try doing is to replace the message statements with the following loop, which prints _every_ variable:

```
 message(STATUS "=============================================================")
 get_cmake_property(_variableNames VARIABLES)
 foreach (_variableName ${_variableNames})
     message(STATUS "${_variableName}=${${_variableName}}")
 endforeach()
 message(STATUS "=============================================================")
```
Search through the variables to see which ones relate to `octomap`.

Regarding an actual resolution of this issue, let's wait for @mwoehlke-kitware or @jamiesnape for advice.

MoyinOtubela, 2016-10-16:
I edited the file again as given in the advice.

[image]

 All variables relating to octomap are in the figure below ( I think ). If there's more that I did not capture, I copied and pasted the output for all variables that came out that may be of interest to you in this same comment... Just keep scrolling till the second "========".

[image]

-- octomap_INCLUDE_DIRS = 
-- octomap_INCLUDES = 
-- octomap_LIBRARIES = 
-- octomap_LIBS = 
-- =============================================================
-- AVL_EXECUTABLE=AVL_EXECUTABLE-NOTFOUND
-- BLAS_goto2_LIBRARY=BLAS_goto2_LIBRARY-NOTFOUND
-- BLAS_openblas_LIBRARY=/usr/lib/libopenblas.so
-- BLAS_openblas_WORKS=1
-- BUILDNAME=Linux-g++-4.9
-- BUILD_NAME_SYSTEM_NAME=Linux
-- BUILD_SHARED_LIBS=
-- BUILD_TESTING=ON
-- BULLET_COLLISION_LIBRARY=/home/moyin/dev/drake-distro/build/install/lib/libBulletCollision.so
-- BULLET_COLLISION_LIBRARY_DEBUG=BULLET_COLLISION_LIBRARY_DEBUG-NOTFOUND
-- BULLET_DYNAMICS_LIBRARY=/home/moyin/dev/drake-distro/build/install/lib/libBulletDynamics.so
-- BULLET_DYNAMICS_LIBRARY_DEBUG=BULLET_DYNAMICS_LIBRARY_DEBUG-NOTFOUND
-- BULLET_INCLUDE_DIR=/home/moyin/dev/drake-distro/build/install/include/bullet
-- BULLET_MATH_LIBRARY=/home/moyin/dev/drake-distro/build/install/lib/libLinearMath.so
-- BULLET_MATH_LIBRARY_DEBUG=BULLET_MATH_LIBRARY_DEBUG-NOTFOUND
-- BULLET_SOFTBODY_LIBRARY=/home/moyin/dev/drake-distro/build/install/lib/libBulletSoftBody.so
-- BULLET_SOFTBODY_LIBRARY_DEBUG=BULLET_SOFTBODY_LIBRARY_DEBUG-NOTFOUND
-- BZRCOMMAND=/usr/bin/bzr
-- BZR_UPDATE_OPTIONS=
-- CMAKE_AR=/usr/bin/ar
-- CMAKE_AR=/usr/bin/ar
-- CMAKE_BASE_NAME=g++
-- CMAKE_BINARY_DIR=/home/moyin/dev/drake-distro/build/drake
-- CMAKE_BUILD_TOOL=/usr/bin/make
-- CMAKE_BUILD_TYPE=Release
-- CMAKE_C11_COMPILE_FEATURES=c_static_assert
-- CMAKE_C11_EXTENSION_COMPILE_OPTION=-std=gnu11
-- CMAKE_C11_STANDARD_COMPILE_OPTION=-std=c11
-- CMAKE_C90_COMPILE_FEATURES=c_function_prototypes
-- CMAKE_C90_EXTENSION_COMPILE_OPTION=-std=gnu90
-- CMAKE_C90_STANDARD_COMPILE_OPTION=-std=c90
-- CMAKE_C99_COMPILE_FEATURES=c_restrict;c_variadic_macros
-- CMAKE_C99_EXTENSION_COMPILE_OPTION=-std=gnu99
-- CMAKE_C99_STANDARD_COMPILE_OPTION=-std=c99
-- CMAKE_CACHEFILE_DIR=/home/moyin/dev/drake-distro/build/drake
-- CMAKE_CACHE_MAJOR_VERSION=3
-- CMAKE_CACHE_MINOR_VERSION=7
-- CMAKE_CACHE_PATCH_VERSION=0
-- CMAKE_CFG_INTDIR=.
-- CMAKE_COLOR_MAKEFILE=ON
-- CMAKE_COMMAND=/home/moyin/tools/cmake-3.7.0-rc1/bin/cmake
-- CMAKE_COMPILER_IS_GNUCC=1
-- CMAKE_COMPILER_IS_GNUCXX=1
-- CMAKE_CPACK_COMMAND=/home/moyin/tools/cmake-3.7.0-rc1/bin/cpack
-- CMAKE_CROSSCOMPILING=FALSE
-- CMAKE_CTEST_COMMAND=/home/moyin/tools/cmake-3.7.0-rc1/bin/ctest
-- CMAKE_CURRENT_BINARY_DIR=/home/moyin/dev/drake-distro/build/drake/systems/robotInterfaces
-- CMAKE_CURRENT_LIST_DIR=/home/moyin/dev/drake-distro/drake/systems/robotInterfaces
-- CMAKE_CURRENT_LIST_FILE=/home/moyin/dev/drake-distro/drake/systems/robotInterfaces/CMakeLists.txt
-- CMAKE_CURRENT_SOURCE_DIR=/home/moyin/dev/drake-distro/drake/systems/robotInterfaces
-- CMAKE_CXX11_COMPILE_FEATURES=cxx_alias_templates;cxx_alignas;cxx_alignof;cxx_attributes;cxx_auto_type;cxx_constexpr;cxx_decltype;cxx_decltype_incomplete_return_types;cxx_default_function_template_args;cxx_defaulted_functions;cxx_defaulted_move_initializers;cxx_delegating_constructors;cxx_deleted_functions;cxx_enum_forward_declarations;cxx_explicit_conversions;cxx_extended_friend_declarations;cxx_extern_templates;cxx_final;cxx_func_identifier;cxx_generalized_initializers;cxx_inheriting_constructors;cxx_inline_namespaces;cxx_lambdas;cxx_local_type_template_args;cxx_long_long_type;cxx_noexcept;cxx_nonstatic_member_init;cxx_nullptr;cxx_override;cxx_range_for;cxx_raw_string_literals;cxx_reference_qualified_functions;cxx_right_angle_brackets;cxx_rvalue_references;cxx_sizeof_member;cxx_static_assert;cxx_strong_enums;cxx_thread_local;cxx_trailing_return_types;cxx_unicode_literals;cxx_uniform_initialization;cxx_unrestricted_unions;cxx_user_literals;cxx_variadic_macros;cxx_variadic_templates
-- CMAKE_CXX11_EXTENSION_COMPILE_OPTION=-std=gnu++11
-- CMAKE_CXX11_STANDARD_COMPILE_OPTION=-std=c++11
-- CMAKE_CXX14_COMPILE_FEATURES=cxx_attribute_deprecated;cxx_binary_literals;cxx_contextual_conversions;cxx_decltype_auto;cxx_digit_separators;cxx_generic_lambdas;cxx_lambda_init_captures;cxx_return_type_deduction
-- CMAKE_CXX14_EXTENSION_COMPILE_OPTION=-std=gnu++14
-- CMAKE_CXX14_STANDARD_COMPILE_OPTION=-std=c++14
-- CMAKE_CXX98_COMPILE_FEATURES=cxx_template_template_parameters
[... 848 lines cut ...]
-- testFrameParsermex_LIB_DEPENDS=general;liblast;general;rbmMexFunctions;
-- testGeometryConversionFunctionsmex_LIB_DEPENDS=general;liblast;general;drakeGeometryUtil;general;drakeMexUtil;
-- testGeometryGradientsmex_LIB_DEPENDS=general;liblast;general;drakeGeometryUtil;general;drakeMexUtil;
-- testIKoptionsmex_LIB_DEPENDS=general;liblast;general;drakeRBM;general;drakeMexUtil;general;drakeIKoptions;
-- testMultipleTimeKinCnstmex_LIB_DEPENDS=general;liblast;general;drakeRBM;general;drakeMexUtil;general;drakeRigidBodyConstraint;general;drakeConstructRigidBodyConstraint;
-- testMultipleTimeLinearPostureConstraintmex_LIB_DEPENDS=general;liblast;general;drakeRBM;general;drakeMexUtil;general;drakeRigidBodyConstraint;general;drakeConstructRigidBodyConstraint;
-- testPostureConstraintmex_LIB_DEPENDS=general;liblast;general;drakeRBM;general;drakeMexUtil;general;drakeRigidBodyConstraint;general;drakeConstructRigidBodyConstraint;
[... cut ...]
Java(TM) SE Runtime Environment (build 1.8.0_101-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.101-b13, mixed mode)
-- velocityToPositionDotMappingmex_LIB_DEPENDS=general;liblast;general;rbmMexFunctions;
-- xfoil_DIR=xfoil_DIR-NOTFOUND
-- xvfb-run=xvfb-run-NOTFOUND
-- yaml-cpp_DIR=/home/moyin/dev/drake-distro/build/install/lib/cmake/yaml-cpp
-- =============================================================
-- Not running systems/robotInterfaces/calibration/test/testJointCalibration because large tests are not enabled

liangfok, 2016-10-17:
@MoyinOtubela, thanks! Can you also re-run the build with ROS sourced? I'm particularly interested in the values of the following variables:

```
-- OCTOMAP_INCLUDE_DIRS
-- OCTOMAP_LIBRARIES
-- OCTOMAP_LIBRARY_DIRS
-- OCTOMAP_MAJOR_VERSION
-- OCTOMAP_MINOR_VERSION
-- OCTOMAP_PATCH_VERSION
-- OCTOMAP_VERSION
```

MoyinOtubela, 2016-10-17:
I did as you asked and the results are below. As expected, the compile error reappeared as seen before with ROS sourced. 

[image]
[image]

liangfok, 2016-10-17:
Thanks. It's so strange how the variables are identical in both the successful and failing case! There must be some other difference that we're not detecting yet.

MoyinOtubela, 2016-10-17:
I was thinking the same thing.

liangfok, 2016-10-17:
Another guess is there might be a difference between the `LD_LIBRARY_PATH` environment variable when ROS is sourced versus when it is not sourced. Perhaps compare the output of the following command with and without ROS:

```
$ echo $LD_LIBRARY_PATH
```

MoyinOtubela, 2016-10-17:
with ROS sourced:
moyin:build$ echo $LD_LIBRARY_PATH
/opt/ros/indigo/lib:/opt/ros/indigo/lib/x86_64-linux-gnu::/home/moyin/Downloads/gurobi652/linux64/lib

without ROS sourced:
moyin:~$ echo $LD_LIBRARY_PATH
:/home/moyin/Downloads/gurobi652/linux64/lib

liangfok, 2016-10-17:
Thanks. Now I'm curious whether we can prioritize the version of `octomap` in Drake over the one in ROS. Try doing the following:

Source ROS.

Modify `LD_LIBRARY_PATH` to prioritize Drake's version of `octomap`:

```
$ cd drake-distro/externals/octomap/octomap/lib
$ export LD_LIBRARY_PATH=`pwd`:$LD_LIBRARY_PATH
```

Proceed to build Drake. See if the linker error continues to exist.

MoyinOtubela, 2016-10-17:
Yes that worked! The output is below.

[image]
[image]
[image]
[image]
Thanks!

MoyinOtubela, 2016-10-17:
Those show before and after results by the way.

liangfok, 2016-10-17:
Great! Thanks for running the tests. We now know of an alternative workaround that does not require disabling ROS Indigo.

MoyinOtubela, 2016-10-18:
I just realized that I didn't run your tests correctly and the error is still present. I somehow forgot to source ROS in one of the steps. You can notice this by examining my earlier screenshot. There was no change to the LD_LIBRARY_PATH to ROS as printed earlier:

"
with ROS sourced:
moyin:build$ echo $LD_LIBRARY_PATH
/opt/ros/indigo/lib:/opt/ros/indigo/lib/x86_64-linux-gnu::/home/moyin/Downloads/gurobi652/linux64/lib

without ROS sourced:
moyin:~$ echo $LD_LIBRARY_PATH
:/home/moyin/Downloads/gurobi652/linux64/lib
"
The output for "echo $LD_LIBRARY_PATH" should be different and it's as illustrated below.

[image]

Sorry about this. Please correct me if I'm wrong.

jamiesnape, 2016-10-18:
IIRC, the problem is the wrong header file rather than the wrong library. We will investigate.

Triton-11, 2016-11-03:
Hey guys, any update on this issue?
I had the exact same problem. I had drake working fine (MATLAB enabled). Then I installed ROS, tried do redo a clean drake superbuild and it failed because of the "Undefined Reference to createChild()" issue. 
As suggested, I "unsourced" ROS in the bashrc file and the build was running fine again.
(Ubuntu 14.04.5 LTS, MATLAB R2015b 8.6.0.267246, ROS indigo, building with ninja)
Is there solution coming for this?
Thanks!

jamiesnape, 2016-11-03:
Not yet, I am afraid.

liangfok, 2016-12-29:
I believe a quick workaround is _not_ sourcing your ROS workspace prior to building Drake (by "sourcing" I mean executing `source $HOME/dev/drake_catkin_workspace/devel/setup.bash` or ` source /opt/ros/indigo/setup.bash`). Once Drake is built, you can source your ROS workspace.

liangfok, 2017-01-06:
I'm going to bump the priority of this because I just learned of another TRI researcher encountering this problem.

liangfok, 2017-01-06:
BTW, the above-mentioned problem was also encountered when compiling Drake using `catkin`. This, I believe, is a new issue. However, I'll withhold creating a new issue until I'm certain both can't be resolved simultaneously.

liangfok, 2017-01-07:
Here is the error I'm observing when building Drake using `catkin` on Ubuntu 14.04 + ROS Indigo:

```
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/src/geometry/octree/octree.cpp:38:
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:179:16: error: no member named 'getNodeChild' in 'octomap::OcTree'
  return tree->getNodeChild(node, childIdx);
         ~~~~  ^
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:191:16: error: no member named 'getNodeChild' in 'octomap::OcTree'
  return tree->getNodeChild(node, childIdx);
         ~~~~  ^
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:203:16: error: no member named 'nodeChildExists' in 'octomap::OcTree'
  return tree->nodeChildExists(node, childIdx);
         ~~~~  ^
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:214:16: error: no member named 'nodeHasChildren' in 'octomap::OcTree'
  return tree->nodeHasChildren(node);
         ~~~~  ^
```
When building `drake/externals/fcl`, the build system is apparently finding ROS Indigo's older version of `octomap` rather than Drake's newer version. ROS' version of `octomap` is part of Debian package `ros-indigo-octomap`:

```
$ dpkg -l | grep octomap
ii  ros-indigo-octomap                                    1.6.9-0trusty-20160321-174526-0700                  amd64        The OctoMap library implements a 3D occupancy grid mapping approach, providing data structures and mapping algorithms in C++. The map implementation is based on an octree. See http://octomap.github.io for details.
ii  ros-indigo-octomap-msgs                               0.3.3-0trusty-20160627-171624-0700                  amd64        This package provides messages and serializations / conversion for the OctoMap library.
```

Drake's version of `octomap` is in `drake-distro/externals/octomap`.

FCL's build system is searching for `octomap` using the [following logic](https://github.com/flexible-collision-library/fcl/blob/06d48b3b6f3605b8caf119d5208d8156eb64fe0d/CMakeLists.txt#L143):

```
  find_package(octomap QUIET)
  # octomap-config.cmake may not define OCTOMAP_VERSION so fall back to
  # pkgconfig
  if(NOT OCTOMAP_VERSION AND PKG_CONFIG_FOUND)
      pkg_check_modules(OCTOMAP QUIET octomap)
  endif()
```

liangfok, 2017-01-07:
Another idea is to disable Drake's version of FCL and Octomap, which will cause the build system to use ROS' version. To do this, I use the following configure command:

```
$ catkin config --cmake-args -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DWITH_DREAL=FALSE -DDISABLE_MATLAB=TRUE -DWITH_OCTOMAP=FALSE -DWITH_FCL=FALSE
$ catkin build --verbose -j1
```

Unfortunately, the build fails because ROS' `ros-indigo-fcl` Debian package does not include `<fcl/fcl.h>`, which is used by `drake-distro/drake/multibody/collision/fcl_model.cc`. This might be another reason do not use `<fcl/fcl.h>`. It's a header file that includes a whole bunch of other header files.

liangfok, 2017-01-07:
I posted the following question to see if anybody has any intuition why executing a compiler command via Catkin would differ from executing the same command directly in the command line:

http://answers.ros.org/question/251449/compiler-behavior-differs-when-executed-by-catkin-vs-command-line/

liangfok, 2017-01-08:
Switching from using:

```
-isystem /home/liang/dev/drake_catkin_workspace/install/include
```

to:

```
-I/home/liang/dev/drake_catkin_workspace/install/include
```

results in a successful command-line build even when `CPATH=/opt/ros/indigo/include`.

Apparently, `-isystem` is lower priority than `CPATH`. I have not tested whether this works using `catkin`.

liangfok, 2017-01-08:
Adding the following flag also works when executing the build command in the command line:

```
-I/home/liang/dev/drake_catkin_workspace/src/drake/externals/octomap/octomap/include
```

I was inspired to try this because Drake's FCL's header files were included in a similar manner:

```
-I/home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl/include
-I/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include
```

I'll next see whether I can get Catkin to include the `-I.../octomap/octomap/include` parameter.

liangfok, 2017-01-09:
I attempted to prioritize Drake's version of FCL and Octomap by using the `CMAKE_CXX_FLAGS` flag. For example:

```
$ cd ~/dev/drake_catkin_workspace
$ catkin config --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DWITH_DREAL=FALSE -DDISABLE_MATLAB=TRUE -DCMAKE_CXX_FLAGS="-I`pwd`/src/drake/externals/octomap/octomap/include -I`pwd`/install/include"
```

Unfortunately, I couldn't get it to work after trying various combination of paths in the set [src/, build/, install/]. For example, one error was:

```
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/drake/multibody/collision/test/fcl_model_test.cc:3:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/../../install/include/fcl/fcl.h:62:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/../../install/include/fcl/geometry/octree/octree.h:145:
/home/liang/dev/drake_catkin_workspace/src/drake/../../install/include/fcl/geometry/octree/octree-inl.h:178:5: error: function-like macro 'OCTOMAP_VERSION_AT_LEAST' is not defined
#if OCTOMAP_VERSION_AT_LEAST(1,8,0)
```

liangfok, 2017-01-09:
I was able to get Drake to build using `catkin` by uninstalling `ros-indigo-octomap` and `ros-indigo-fcl`:

```
$ sudo apt-get install ros-indigo-octomap ros-indigo-fcl
$ cd ~/dev/drake_catkin_workspace
$ source /opt/ros/indigo/setup.bash
$ catkin config --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DWITH_DREAL=FALSE -DDISABLE_MATLAB=TRUE
$ catkin build --verbose -j1
```

Since the above command uses only one thread (`-j1`), it took a _very long time_ to build, but at least it was successful, and I assume it will also work using a multi-threaded build.

```
[build] Summary: All 7 packages succeeded!                                                                                                                                                                                            
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                                                                                      
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                                                                              
[build]   Abandoned: None.                                                                                                                                                                                                            
[build]   Failed:    None.                                                                                                                                                                                                            
[build] Runtime: 1 hour 14 minutes and 34.5 seconds total.  
```

PR #4709 (2017-01-09): Updates ROS Indigo Installation Instructions to Remove Octomap and FCL
This PR resolves #3814 the brute-force way: by removing ROS' versions of Octomap and FCL. This was done as a last resort. See notes in #3814 on various ways we tried to get Drake to prioritize its own versions of Octomap and FCL libraries over ROS' versions.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4709)
closed 2017-01-09 by liangfok