RobotLocomotion/drake #3818: Gurobi QP is still buggy?

siyuanfeng-tri, 2016-10-17:
Here is a random problem I generated in matlab. I tested the current gurobi solver against the snopt solver, and the answers are different. The snopt answer matches matlab's qp. 

unit test code: 
```
GTEST_TEST(testGurobi, checkGurobiQPAgainstSnopt) {
  MathematicalProgram prog;
  auto x = prog.AddContinuousVariables(2, "x");
  auto z = prog.AddContinuousVariables(2, "z");
  auto y = prog.AddContinuousVariables(2, "y");

  Eigen::Matrix<double, 2, 4> CE_xy;
  CE_xy << 0.3147,    0.1324,    0.4575,    0.4572,
        0.4058,   -0.4025,    0.4649,   -0.0146;
  Eigen::Matrix<double, 2, 1> ce0_xy;
  ce0_xy << -0.0782,
         0.4157;

  Eigen::Matrix<double, 1, 4> CI_xy;
  CI_xy << 0.1797,    0.0853,    0.4593,    0.3143;
  Eigen::Matrix<double, 1, 1> ci_xy_upper, ci_xy_lower;
  ci_xy_upper << -10;
  ci_xy_lower << -INFINITY;

  Eigen::Matrix<double, 1, 2> CI_z;
  CI_z << -0.3639,    0.3693;
  Eigen::Matrix<double, 1, 1> ci_z_upper, ci_z_lower;
  ci_z_upper << -10;
  ci_z_lower << -INFINITY;

  Eigen::Matrix<double, 6, 6> G;
  G <<
    0.6472,   -0.1517,   -0.1019,   -0.4394,    0.1772,    0.1697,
    -0.1517,    0.3755,    0.0967,    0.3431,    0.2979,    0.1124,
    -0.1019,    0.0967,    0.1902,    0.0987,    0.2064,   -0.0820,
    -0.4394,    0.3431,    0.0987,    0.6460,    0.1524,    0.0546,
    0.1772,    0.2979,    0.2064,    0.1524,    1.0408,    0.2175,
    0.1697,    0.1124,   -0.0820,    0.0546,    0.2175,    0.5526;

  Eigen::Matrix<double, 6, 1> g0;
  g0 << 0.3001,
     -0.0686,
     0.4106,
     -0.3182,
     -0.2362,
     -0.3545;

  prog.AddQuadraticCost(G, g0, {x, z, y});
  prog.AddLinearEqualityConstraint(CE_xy, ce0_xy, {x, y});
  prog.AddLinearConstraint(CI_xy, ci_xy_lower, ci_xy_upper, {x, y});
  prog.AddLinearConstraint(CI_z, ci_z_lower, ci_z_upper, {z});

  SolutionResult result;
  SnoptSolver snopt;
  result = snopt.Solve(prog);
  if (result != drake::solvers::SolutionResult::kSolutionFound) {
    std::cerr << "solution not found\n";
  }
  Eigen::VectorXd snopt_solution = prog.GetSolutionVectorValues();

  GurobiSolver gurobi;
  result = gurobi.Solve(prog);
  if (result != drake::solvers::SolutionResult::kSolutionFound) {
    std::cerr << "solution not found\n";
  }
  Eigen::VectorXd gurobi_solution = prog.GetSolutionVectorValues();

  std::cout << "gurobi: " << gurobi_solution << " snopt " << snopt_solution << std::endl;
  EXPECT_TRUE(
      CompareMatrices(gurobi_solution, snopt_solution, 1e-8, MatrixCompareType::absolute));
}
```

And matlab code:
```
G = [    0.6472,   -0.1517,   -0.1019,   -0.4394,    0.1772,    0.1697;
    -0.1517,    0.3755,    0.0967,    0.3431,    0.2979,    0.1124;
    -0.1019,    0.0967,    0.1902,    0.0987,    0.2064,   -0.0820;
    -0.4394,    0.3431,    0.0987,    0.6460,    0.1524,    0.0546;
    0.1772,    0.2979,    0.2064,    0.1524,    1.0408,    0.2175;
    0.1697,    0.1124,   -0.0820,    0.0546,    0.2175,    0.5526];
g0 = [0.3001;
     -0.0686;
     0.4106;
     -0.3182;
     -0.2362;
     -0.3545];
 
CE = [0.3147,    0.1324,    0, 0, 0.4575,    0.4572;
        0.4058,   -0.4025,    0, 0, 0.4649,   -0.0146];
    
ce0 = [-0.0782;
         0.4157];
     
     
CI = [0.1797,    0.0853,    0, 0, 0.4593,    0.3143;
      0, 0, -0.3639,    0.3693, 0, 0];
ci0 = [-10;
    -10];

X = quadprog(G, g0, CI, ci0', CE, ce0')
% X =
% 
%    65.2807
%     4.5236
%    70.6163
%    42.5054
%   -51.9950
%     5.6141
```

hongkai-dai, 2016-10-17:
I tested your program, actually I think the problem is that gurobi solution and snopt solution does not match so well. Their difference is in the order of 1E-5. If you inspect the gurobi solution, they also math the MATLAB solution, up to 4th decimal point.

hongkai-dai, 2016-10-17:
The test is done on my branch https://github.com/hongkai-dai/drake/tree/gurobi_socp, in which I continued to fix problem in gurobi. The current master is still buggy. I will send the PR hopefully today.

hongkai-dai, 2016-10-17:
The problem is that in the `gurobi_solver` on master branch, it does not allow to impose constraints on a subset of variables, like 
```
prog.AddLinearEqualityConstraint(CE_xy, ce0_xy, {x, y});
```
This problem is going to fix in my next PR, on this branch https://github.com/hongkai-dai/drake/tree/gurobi_socp

Thanks for the testing program! I will also add this test to the PR.

siyuanfeng-tri, 2016-10-18:
Great! I saw your other triplet pr went it, and thought that might have fixed this.

PR #3834 (2016-10-22): Gurobi socp
This PR includes following features.

1. Gurobi can now solve second order conic programming problem.
2. Gurobi can add constraints on a subset of variables. Resolves #3818 
3. MathematicalProgram stores `linear_costs_`. This cost is going to be used for all conic programming problem, since the standard form of conic programming is to solve a linear objective.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3834)
closed 2016-10-22 by hongkai-dai