RobotLocomotion/drake #3837: MEX debugging: gdb segfault on read from JNA deleted shared library

samkhal, 2016-10-18:
I'm getting a mex error running my code, and I'm trying to debug it. However, gdb is crashing due to, as far as I understand, java JNA libraries that are created and deleted at runtime. Here's my debugging process; anybody know how I can resolve this? 

I set the build type to "Debug" with `ccmake`.
```
$ matlab -Dgdb

(gdb) handle SIGSEGV SIGBUS nostop noprint
(gdb) set print thread-events off
(gdb) break constructQPDataPointerMex.cpp:7
(gdb) run -nodesktop -nosplash

>> dbmex on
>> runAtlasWalking

BFD: reopening /tmp/jna-113631/jna7782802719441819747.tmp: No such file or directory

BFD: reopening /tmp/jna-113631/jna7782802719441819747.tmp: No such file or directory

Can't read data for section '.eh_frame' in file '/tmp/jna-113631/jna7782802719441819747.tmp'
[Switching to Thread 0x7fffdb4bb700 (LWP 9327)]

Thread 3 "MATLAB" hit Breakpoint 2, mexFunction (nlhs=<unavailable>, plhs=<unavailable>, 
    nrhs=<unavailable>, prhs=<unavailable>)
    at /home/sam/drake-distro/drake/systems/controllers/constructQPDataPointerMex.cpp:7
7	void mexFunction(int nlhs, mxArray *plhs[], int nrhs, const mxArray *prhs[]) {
(gdb) continue
Continuing.
Segmentation fault (core dumped)
```

The segfaults are irregular and unexpected. I assume running with `-nojvm` would fix the issue, but lots of drake pieces rely on the jvm; how do I get around this?

RussTedrake, 2016-10-19:
debugging mex files with gdb is a nightmare (not our fault).  we've made it work with gdb rules which ignore certain "segfaults" on some versions of matlab/java, but have considered it impossible with others.  life is better in our c++ only workflow.

closed 2016-10-19 by RussTedrake

samkhal, 2016-10-19:
Sigh. Looking forward to the port.