RobotLocomotion/drake #3841: Translator for bot_core.robot_state_t Needed

liangfok, 2016-10-19:
We need a translator that converts the kinematics result outputted by `RigidBodyPlant` into a `bot_core::robot_state_t` message. This will enable the robot's state to be published on an LCM channel, which would then allow for the creation of interactive apps.

liangfok, 2016-10-19:
This issue depends on a resolution of #3598.

sammy-tri, 2016-10-19:
Does it depend on #3598 for a minimal implementation?  I wrote a simple System1 example which I only ever tested on the iiwa (and which lacked velocities and efforts), but that was sufficient for the amount of director interoperability I needed for basic planning.

sammy-tri, 2016-10-19:
Example referenced above:

https://github.com/sammy-tri/drake/blob/director_ik/drake/examples/kuka_iiwa_arm/robot_state_publisher.h

liangfok, 2016-10-19:
It depends on whether we want to reuse the existing `LcmPublisherSystem` and `LcmSubscriberSystem` systems. If we do, we need to wait for #3598 since that supports `AbstractValue` types. If we do not, then you're right, we do not need to wait for #3598.

sammy-tri, 2016-10-19:
The example above populates the message from the vector valued output of RigidBodySystem.  I believe the vector valued output on port 0 of RigidBodyPlant is basically the same, so we would still be able to use LcmPublisherSystem and LcmSubscriberSystem?

(ignoring the parts not populated in my example)

liangfok, 2016-10-19:
Oh yes, you're right. If we decide to use the generalized state output from the `RigidBodyPlant`, we do not need to block on #3598. We only need to block on #3598 if we decide to use the kinematics results output of `RigidBodyPlant`.

liangfok, 2016-10-19:
If we decide to use the generalized state output from `RigidBodyTree`, the translator will need to maintain a pointer to the `RigidBodyTree` so it can re-derive the body poses. Is this OK?

sammy-tri, 2016-10-19:
It's consistent with RigidBodyTreeLcmPublisher, and I don't personally object...  (the latter being neither a necessary nor sufficient condition)

liangfok, 2016-10-19:
Yes, it would be consistent with [ViewerDrawTranslator](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/rigid_body_plant/viewer_draw_translator.h#L33).

Looking more closely at [bot_core::robot_state_t](https://github.com/mwoehlke-kitware/bot_core_lcmtypes/blob/master/lcmtypes/bot_core_robot_state_t.lcm), I'm now wondering what the "pose" and "twist" values are for. Is the floating joint not represented in the list of joint states?

sammy-tri, 2017-12-11:
This was effectively fixed by #6073

closed 2017-12-11 by sammy-tri