RobotLocomotion/drake #3852: Mysterious CI Error - Polynomial<double>("!") Failed To Throw Exception

liangfok, 2016-10-20:
# The Error

```
/home/ubuntu/workspace/linux-clang-ninja-continuous-debug/drake/common/test/polynomial_test.cc:218: Failure
Expected: Polynomial<double>("!") throws an exception of type std::runtime_error.
  Actual: it throws nothing.
```

# Example CI Test Logs

* https://drake-jenkins.csail.mit.edu/job/linux-clang-ninja-continuous-debug/797/

# Prognosis

Maybe there's something stochastic in this test? @jwnimmer-tri might have experienced it months ago. Not sure.

liangfok, 2016-10-20:
I'm going to remove the CI label since I don't think this is specific to CI. I'll also update the title.

jwnimmer-tri, 2016-10-20:
The test case calls the `Polynomial` constructor with a string argument of `!`.  This calls `VariableNameToId`, which calls `strchr` to find the index of `!`, which is not found so returns `nullptr`, which we then compute the distance from `kNameChars ` to `nullptr`, which is random based on the linker.  The DUT should directly raise an exception on an invalid name, instead of relying on the math mapping into int-space to go out of range.

jwnimmer-tri, 2016-10-20:
This is a fun one too, because it's not UB, just a bug -- so valgrind won't find it.

liangfok, 2016-10-20:
Awesome. Thanks for finding the bug! What does "UB" stand for?

PR #3877 (2016-10-22): Avoid huge-integer accidents on invalid input
Closes #3852.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3877)
closed 2016-10-22 by jwnimmer-tri