RobotLocomotion/drake #3855: fastqp_solver_test is testing snopt

hongkai-dai, 2016-10-20:
@RussTedrake , when I run `fastqp_solver_test.cc`, it actually runs SNOPT solver, at line 59
``` 
result = prog.Solve();  // TODO(russt) call fastQP solver
                              // explicitly
```
Could  you elaborate why you did not create a `FastqpSolver` as `SnoptSolver`, and explicitly call fastqp solver?

RussTedrake, 2016-10-20:
Because i want it to call gurobi when that is present!  :)

hongkai-dai, 2016-10-20:
Excellent! I am turning on gurobi solver for this test. 

But it does not run fastqp currently, instead it runs SNOPT, at least when I tested it on my local machine. Since there is not a branch to run fastqp in `MathematicalProgram::Solve`.

closed 2016-10-22 by hongkai-dai

RussTedrake, 2016-10-25:
Correct. That's still todo. And also depends on being able to extract the
active set information after the solve.