RobotLocomotion/drake #3865: Factor out a `KukaIiwaPlant` from `KukaIiwaArmDynamicsSim` and `kuka_demo.cc`

liangfok, 2016-10-21:
For context, see: https://github.com/RobotLocomotion/drake/pull/3745#issuecomment-255381050

sammy-tri, 2016-10-21:
I've got a 3rd (at least?) version of the mostly-duplicated code in one of my branches too.  Not sure how close I am to a PR, but I'll avoid worrying about de-duplicating locally if you're working on a generic version already.

liangfok, 2016-10-21:
Yes, please go ahead and proceed with your 3rd duplicate. We'll factor out a `KukaIiwaPlant` from all three once we get around to addressing this issue.

sammy-tri, 2016-12-27:
Reassigning to @naveenoid since I think #4521 may accomplish most/all of this.

naveenoid, 2016-12-27:
Since #4521 has landed the only thing needed is to get KukaIiwaArmDynamicsSim to use the plant model. I will submit a fix PR ASAP.

naveenoid, 2017-05-11:
Closing this issue in light of the creation of drake/manipulation.

closed 2017-05-11 by naveenoid