RobotLocomotion/drake #3867: Systems/Observers need porting over to Sys2.0 C++

naveenoid, 2016-10-21:
This is to start discussions on how to effectively port over the observers into Sys2.0 C++. I would like to take a crack at the LuenbergerObserver.m first especially given that Affine/LinearSystem has nearly landed. If any one else has ideas please note here.

hongkai-dai, 2016-10-22:
LuenbergerObserver sounds a great start. What is the example you have in mind?

naveenoid, 2016-10-23:
I did not have any clear example in mind except for sample problems in linear observer design. Any recommendations? I personally want to work towards KF/EKF..I figured this would be a perfect first step.

jwnimmer-tri, 2016-12-22:
https://github.com/RobotLocomotion/drake/tree/master/drake/systems/estimators already has lots.  Update the issue with next steps, or close it?

RussTedrake, 2016-12-23:
i think it's safe to close.

closed 2016-12-23 by RussTedrake