RobotLocomotion/drake #3873: Provide valgrind pre-merge build, developer tooling, and documentation

david-german-tri, 2016-10-21:
- [ ] Prevent valgrind from traversing MATLAB and Python.
 - [ ] In CI
 - [ ] In local builds

- [ ] Set the environment variable `GTEST_DEATH_TEST_USE_FORK` for the memcheck build.

- [ ] Audit drake/valgrind.supp

- [ ] Document how developers should invoke the memcheck build locally.
 - [ ] Should not use a log file or `-q`

jamiesnape, 2016-10-21:
> Prevent ctest from generating --log-file
Not possible as far as I am aware. CTest needs to be able to parse it.

david-german-tri, 2016-10-21:
It looks like as long as `-q` is absent, `--log-file` is not a problem.  Updated the checklist accordingly.

jamiesnape, 2016-10-21:
For developers, what do you want me to do about compiler flags? We basically run `-g -O1 -fno-omit-frame-pointer` on CI. I can add a CMake option that set the flags, simply document the flags, or allow developers to make their own decisions on flags and interpreting the output accordingly.

jamiesnape, 2016-10-21:
`-O1` is not entirely necessary, of course, but it makes things more bearable.

david-german-tri, 2016-10-21:
I think an option is the way to go, where the default matches CI.

david-german-tri, 2016-10-24:
I'm thinking we will need to exclude some tests from memcheck (notably, those that invoke optimizers) because they're just too slow.  Is there a recommended phrasing for that in CMake?

jwnimmer-tri, 2016-10-24:
If you get the test `SIZE` set appropriately, it will automatically be run either pre-merge and/or continuous and/or nightly (and/or useful for developers who want to filter-by-size as well).

jamiesnape, 2016-10-25:
@david-german-tri Assuming that there are in `enormous` range for sizing, for the moment, add `NO_MEMCHECK` to the end of the `drake_add_*_test` calls. It will be ignored for now, but I can add something as I am fixing up the compiler flags for developers.

david-german-tri, 2016-10-25:
It looks like the memcheck build reports success in Jenkins once all the tests are passing, even if there are memory defects:

https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-linux-clang-nightly-memcheck-valgrind-270&field2=buildstarttime&compare2=84&value2=now

vs 

https://drake-jenkins.csail.mit.edu/view/Nightly/job/linux-clang-nightly-memcheck-valgrind/270/

Adding to checklist in OP.

jamiesnape, 2016-10-25:
Yes, that was deliberate.

david-german-tri, 2016-10-25:
> Yes, that was deliberate.

Why?  Seems like the opposite of what we want.

jamiesnape, 2016-10-25:
It is now, but as there was no move to fix any of the defects it was just noise.

jamiesnape, 2016-10-25:
Actually, I take that back, we are suppressing coverage errors, not memcheck. I will look into it.

jamiesnape, 2016-10-25:
(don't fix that detect for now)

jamiesnape, 2016-10-25:
Apparently `valgrind` passes down the return value of the program that it is simulating unless you pass it an option, and `ctest` is respecting that value. @BetsyMcPhail is looking at a fix.

david-german-tri, 2016-11-22:
https://github.com/RobotLocomotion/drake-ci/pull/45

david-german-tri, 2016-12-19:
#3920 checked off a number of boxes here.

stonier, 2017-08-10:
Right now the valgrind jobs are up and running well, with only the exception that they haven't been moved to Continuous (i.e. PR hooks). With budget constraints we're going to leave them in Nightly and monitor how often they pick up problems. If it turns out to be frequent, then we'll review the decision.

FYI @jwnimmer-tri

closed 2017-08-10 by stonier

jwnimmer-tri, 2017-08-10:
SGTM.  If Nightly breaks too often, we could `--config fastmemcheck` (instead of `--config memcheck`) atop an existing Release-mode pre-merge build, which I think would be pretty cheap.

(I'm totally happy to keep it Nightly for now though, as you say.)