RobotLocomotion/drake #3885: Add open-source builds in pre-merge CI

jwnimmer-tri, 2016-10-24:
We should add open-source builds to pre-merge CI.  As we get increasingly more work on the C++ solvers (optimizers) stack, it is increasingly risky that edits will break when run with a different configuration of solvers that the developer tested with.  Pre-merge currently only captures the "all" and "none" cases; the middle ground of "everything that's unrestricted" can show up in frequent post-merge hassles such as #3883 (and other breaks from the recent past).

How about an open-source non-matlab build as part of the pre-merge CI matrix (and thus also in Continuous Production).

See also #3772.

jwnimmer-tri, 2016-10-24:
(Even if we change the conventional unit-test-nerfing rules (e.g., only run a bunch of unit tests if gurobi and snopt are both installed, don't try to downgrade-to-nlopt/ipopt), we should still cover the open-source configuration in CI.)

jamiesnape, 2016-10-24:
I am starting to think we should have builds with a single solver switched on in each case.

david-german-tri, 2016-10-24:
#3883 was actually MATLAB-specific. I'd be in favor of turning on linux-gcc-experimental-matlab-open-source in pre-merge as an easy first step; then we can see what pain points remain.

sammy-tri, 2016-10-24:
I think builds with individual solvers turned on might be excessive in terms of the amount of time to complete all of the builds.  In the tests which use the MathematicalProgram interface to directly test a solver, each solver is iterated explicitly (and skipped if not enabled in that build).  This is harder in cases like the current failure where the test that failed was a class which uses MathematicalProgram as a backend.  An open-source build in pre-merge CI is probably a good idea, and I'll also give some thoughts as to how to allow use of a wider range of solvers in CI without a ton of builds.

jamiesnape, 2016-10-24:
> I am starting to think we should have builds with a single solver switched on in each case.

I should add, not on pull requests. Personally I think we have plenty on those already. It just depends how sensitive you are to build breakages.

jamiesnape, 2016-10-24:
I have some non-MATLAB open source builds set up now. We should let them run through continuous/nightly for a cycle or two before promoting them maybe sometime Wednesday.

david-german-tri, 2016-11-04:
They've been promoted.

closed 2016-11-04 by david-german-tri