RobotLocomotion/drake #3899: Generalize PidControlledSystem<T> To Support Plants With More Outputs Than Actuator Inputs

liangfok, 2016-10-25:
The current implementation of `PidControlledSystem<T>` requires that the plant being controlled contain exactly twice as many output ports as input ports. This prevents `PidControlledSystem<T>` from working with `RigidBodyPlant` systems of models that have quaternion floating joints or are under-actuated. We need to somehow tell `PidControlledSystem<T>` which of the outputs should be used as part of the feedback control signal.

This issue is preventing the System 2.0 version of `car_sim_lcm` from passing CI in #3886.

jwnimmer-tri, 2016-10-25:
While deeply unsatisfying, I think you could work around this with a Demultiplexer and Multiplexer block for now in the demo?

liangfok, 2016-10-25:
Yeah, I could use a demultiplexer to split all 17 outputs into individual signals, and then use a multiplexer to combine some of them into the 6-vector feedback signal. For generality, I'm thinking about implementing a "selector" system that does this in one step. In this case, it'll take as a parameter a 27x6 selection matrix that selects which of the 27 input signals are outputted.

jwnimmer-tri, 2016-10-25:
What is the simplest thing that could possibly work?

liangfok, 2016-10-25:
I just realized that the "selector" system can be implemented using the `MimoGain` system, which is already part of #3886! I think wiring that into the existing `PidControlledSystem` will be the easiest way to resolve this issue.

liangfok, 2016-10-26:
I have a working prototype solution to this issue using `MimoGain` in the following WIP branch:

https://github.com/liangfok/drake/blob/feature/sys2_car_sim/drake/systems/controllers/pid_controlled_system.cc#L81

jwnimmer-tri, 2016-10-31:
I think this already has a PR, but I don't see it linked here?

liangfok, 2016-10-31:
Sorry. Linking now.

PR #3947 (2016-11-01): Generalizes PidControlledSystem To Support Gain Vector and Plants where # Outputs > 2 * # Inputs
This is necessary for the `car_sim_lcm` demo (towards: #3886).

Closes #3899.

---

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3947)
closed 2016-11-01 by liangfok