RobotLocomotion/drake #3901: Move deprecated System 1.0 infrastructure to drake/system1

jwnimmer-tri, 2016-10-25:
We should physically separate deprecated System 1.0 infrastructure, both so that the new world is cleaner, and so that we can trivially identify in-tree code that's using deprecated APIs (e.g., `rgrep system1/`).

I have a branch ready at https://github.com/jwnimmer-tri/drake/tree/build-system1 with this change, and will PR once #3861 lands.

closed 2016-10-25 by jwnimmer-tri

RussTedrake, 2016-10-25:
why not just zap it?  I think someone told me that “git remembers”.  anybody still using it?

david-german-tri, 2016-10-25:
#3902 suggests there are a dozen or so usages remaining.

jwnimmer-tri, 2016-10-25:
Right.  If I have permission to remove Atlas, Kuka, Pendulum, Automotive-MultiCar, Trajectory Optimization, and IK from the tree, I'm happy to nix System 1.0.  Otherwise, we now have a nice greppable burn-down list of items to port.

PR #3902 (2016-10-25): Move deprecated System 1.0 infrastructure to drake/system1
Fixes #3901.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3902)