RobotLocomotion/drake #3906: Adding constraint/cost using symbolic expressions

hongkai-dai, 2016-10-26:
Based on a meeting with @soonho-tri , @ggould-tri and @sammy-tri , we decide to add initial support for constraint with symbolic expression. I summarize one tentative plan here:

1. Add two functions

   ```
MathematicalProgram::AddCostFromSymbolicExpression(const drake::symbolic::Expression& expression, const drake::symbolic::Variables& symbolic_variables, const VariableList& decision_variables);
MathematicalProgram::AddConstraintFromSymbolicFormula(const drake::symbolic::Formula& formula, const drake::symbolic::Variables& symbolic_variables, const VariableList& decision_variables);
   ```
   which takes symbolic expression/formula, the symbolic variables used in the expression/formula, and the corresponding decision variables matching for the symbolic variables.
2. Add a constructor to `Constraint` class, which can construct a constraint (linear, quadratic, polynomial, etc) from a symbolic expression/formula. @soonho-tri , is it possible to determine the type of the constraint (linear, quadratic, polynomial, etc) from the symbolic expression?
3. Inside `AddConstraintFromSymbolicFormula` and `AddCostFromSymbolicExpression`, call the `Constraint` constructor mentioned in step 2, and then call the `AddConstraint` or `AddCost` function.

@soonho-tri , @ggould-tri , @sammy-tri , does this agree with what you think?

soonho-tri, 2016-10-26:
Thanks @hongkai-dai for the nice summarization. This agrees with what I think.

>  @soonho-tri , is it possible to determine the type of the constraint (linear, quadratic, polynomial, etc) from the symbolic expression?

Yes.

hongkai-dai, 2016-10-29:
@soonho-tri , I am thinking about the API of the symbolic expression. I feel the API I posted above is not very user-friendly. The disadvantage is that when adding constraint, it asks the user to give both the symbolic variables, and also the decision variables; so the user has to book-keeping the pair between these two variables. May I propose the following change?

1. Add a new type

    ```
    typedef std::vector<symbolic::Variable> SymbolicVariableVector;
    ```
2. Add a data member

   ```
  std::unordered_map<symbolic::Variable, DecisionVariableView> symbolic_to_decision_variable_map_;
   ```
 in `MathematicalProgram` class, together with a function to add symbolic decision variable

    ```
    const SymbolicVariableVector MathematicalProgram::AddContinuousSymbolicVariables(size_t num_new_vars, std::string name = "x") {

      DecisionVariableView_ variable_view = AddContinuousVariables(num_new_vars, name);

      // We should call the right constructor for symbolic::Variable. This line should cause compile error.
      SymbolicVariableVector symbolic_vars(num_new_vars);
     
      for (int i = 0; i < num_new_vars; ++i) {
        symbolic_to_decision_variable_map_.insert(std::pair<symbolic::Variable, DecisionVariableView>(symbolic_vars[i], variable_view(i)));
      }
      
      return symbolic_vars;
}
    ```
3. Then we add a constraint with symbolic expression by calling

   ```
MathematicalProgram::AddConstraintFromSymbolicFormula(const drake::symbolic::Formula& formula);
   ```
   Inside this function, it first extracts all the symbolic variables in the formula, and then find the corresponding `DecisionVariableView` objects by querying the map `symbolic_to_decision_variable_map_`, and form the `VariableList` object. It also extracts the linear term or the polynomial coefficients of the formula. Then we can construct the corresponding constraints, and call `AddConstraint` function in `MathematicalProgram`.

@soonho-tri , @ggould-tri , @sammy-tri , what do you think?

soonho-tri, 2016-10-30:
@hongkai-dai, I think it makes sense.

soonho-tri, 2016-10-30:
`const SymbolicVariableVector MathematicalProgram::AddContinuousSymbolicVariables(size_t num_new_vars, std::string name = "x")`

One minor issue: we don't need `const` in the return type here unless you meant `const SymbolicVariableVector&` (`&` is added).

sammy-tri, 2016-10-31:
I also think it makes sense.  Presumably an error would be thrown if the same symbolic variable was added twice, or if we added a formula with an unknown symbolic variable?  (or a formula with other issues which couldn't be expressed to the constructor of the appropriate constraint?)

hongkai-dai, 2016-10-31:
@sammy-tri FWIW, I just implemented the feature in `MathematicalProgram` to handle linear constraint with duplicate variables. It combines the coefficients of the same variables and form a new variable list and the constraint. I will send that PR soon.

rdeits, 2016-11-01:
Is it really necessary to have `FromSymbolicFormula` in the function name? Doesn't the type of the argument (`drake::symbolic::Formula`) imply that? I'm generally in favor of descriptive names, but this just seems redundant.

soonho-tri, 2016-11-02:
@rdeits, I agree. The following should be enough:

```
MathematicalProgram::AddCost(const drake::symbolic::Expression& expression, ...);
MathematicalProgram::AddConstraint(const drake::symbolic::Formula& formula ...);
```

hongkai-dai, 2016-12-07:
close this issue. #4415 is our updated roadmap.

closed 2016-12-07 by hongkai-dai