RobotLocomotion/drake #3910: Create a "Hello World" IIWA Grasping Demo

liangfok, 2016-10-26:
This would serve as a forcing function and mechanism for verifying that our modeling and simulation capabilities are moving in the right direction.

naveenoid, 2016-10-26:
I can also take a pass at this if you guys are immediately occupied.

liangfok, 2016-10-26:
Great! I'm working on something else at the moment and will add you to the assignees list.

sammy-tri, 2016-10-26:
@naveenoid If you could take on getting the initial world put together with the iiwa, the gripper, and an object all in the demo (even if nothing moves on a commanded trajectory), that would be great!

sherm1, 2016-10-26:
Please let @SeanCurtis-TRI know if the current contact code isn't getting the job done!

liangfok, 2016-10-26:
@naveenoid, once you get started, please point us to your WIP branch. Let's try to make this a "micro sprint" where multiple developers cooperate to speed up the creation of the initial PR.

naveenoid, 2016-10-27:
Awesome..Will do!

naveenoid, 2016-10-31:
Updated the issue in anticipation of my first PR (coming soon). Comments welcome!

sammy-tri, 2016-10-31:
Current edited version looks pretty much right to me.  We'll need to figure out if/how we want to split up the work once we get to spiral 2.  Looking forward to the PR!

naveenoid, 2017-02-27:
Final spiral addressed by #5321

naveenoid, 2017-04-04:
This issue is reopened to track the PR train for the monolithic version of the grasping demo.

closed 2017-04-24 by naveenoid