RobotLocomotion/drake #3919: Generalize AutomotiveSimulator To Support Multi-Body Vehicles

liangfok, 2016-10-27:
# Problem Definition

[AutomotiveSimulator](https://github.com/RobotLocomotion/drake/blob/96090268a98d317e2725b3894be6a7cd0e530a43/drake/automotive/automotive_simulator.h) only supports "box" vehicles, i.e., vehicles that consist of a single rigid body. For visualization purposes, we want to be able to load and visualize complex multi-body vehicles that look like actual vehicles.

# Plan

## Spiral 1 - RPY Floating Joint + Zero Position Joints

This spiral assumes all floating joints are of type `kRollPitchYaw` and all other joints are at their zero positions.

## Spiral 2 - RPY Floating Joint + Non-Zero Position Joints.

This spiral builds upon spiral 1 by  allowing the rest of the joints to be in non-zero positions. It continues to assume all floating joints are of type `kRollPitchYaw`.

## Spiral 3 - Any Floating Joint + Non-Zero Position Joints

This spiral builds upon spiral 2 by supporting any type of floating joint.

## Spiral 4 - Stress Test Various Mesh and Texture Formats

Here we verify that the system supports the most popular mesh and material description formats including `.obj`, `.dae`, etc.

maddog-tri, 2016-10-27:
I suggest augmenting the "Spiral N" goals to give them visible feature goals, too --- e.g., "Prius with static wheels" for Spiral 1, "Prius with steerable wheels" for Spiral 2.  (Maybe 3 becomes RPY+multi-DOF joints, with "Semi-trailer truck with steerable wheels and articulated trailer"?)

liangfok, 2016-10-30:
@maddog-tri, great idea. I updated the original post with visible feature goals. PTAL.

jwnimmer-tri, 2016-10-31:
I am hesitant to do some of the later spirals until we have GeometryWorld up and running.  The bespoke glue code to work around it is not worth too much investment.  We can accept glue code for demos we _need_, such as multi-car automotive demos, but we shouldn't add quadcopters and such until we have GeometryWorld.

liangfok, 2016-10-31:
OK SGTM. Is there a PR #, issue #, or some other documentation I can reference to learn more about the aforementioned `GeometryWorld`?

jwnimmer-tri, 2016-10-31:
I refer to you @david-german-tri who can help coordinate locally and/or refer you to the dynamics team.

sherm1, 2016-10-31:
Pinging @SeanCurtis-TRI for `GeometryWorld` commentary. My 2 cents: I would rather call this `SharedEnvironment` since it will ultimately include much more than geometry.

jwnimmer-tri, 2017-03-11:
@liangfok I'm not sure that this issue as currently phrased is helping much.  Is there any work here remaining that's worth capturing before we close this?

liangfok, 2017-03-11:
I would like to wait for the switch to `PoseAggregator` before continuing work on this issue. I do think it's desirable to eventually support visualization of the wheels spinning, front wheels steering, suspension movements, etc., though not in the near term. I shall thus decrease the priority of this issue for now.

jwnimmer-tri, 2017-03-17:
Ah, okay.  From the original post, I didn't realize that the goal was to map SimpleCar-like dynamics into improved visualizations by adjusting more of the links during the visualization step.

jwnimmer-tri, 2018-09-27:
Closing as WONTFIX.

closed 2018-09-27 by jwnimmer-tri