RobotLocomotion/drake #3924: Enable CMake strict mode

david-german-tri, 2016-10-27:
If a variable is undefined, we would like CMake to fail.

david-german-tri, 2016-11-02:
@jamiesnape will take a look today

david-german-tri, 2016-11-03:
The flags are `--warn-uninitialized` and one other, which I can't remember.  @jamiesnape @billhoffman

david-german-tri, 2016-11-03:
Also, we need to upgrade the warnings to errors.

jamiesnape, 2016-11-03:
`--warn-unused-vars`, but, at first glance, it looks like that is introducing some unhelpful noise.

jamiesnape, 2016-11-17:
`--warn-uninitialized` is enabled on CI.

david-german-tri, 2017-01-05:
@jamiesnape @billhoffman Can one of you write down here what it will take to upgrade the warnings to errors?  (My best recollection is that there are upstream CMake changes which have already landed, but have not been released.)

EricCousineau-TRI, 2018-02-08:
Closing, as build system has migrated to Bazel.

jamiesnape, 2018-02-08:
So we still use CMake for MATLAB and we do have warnings, and we will also have acceptance tests that use CMake eventually, so this is still should be on the backlog, I think.

EricCousineau-TRI, 2018-02-08:
Done.

jwnimmer-tri, 2018-11-14:
Eric and I wonder that we don't seem to see many errors related to CMake scripting bugs that these warnings would highlight.  So this is still relevant, but if it's not a pain point we can close this and revisit if/when we have more CMake code?  What do you think @jamiesnape?