RobotLocomotion/drake #3945: Indicator constraints in mixed-integer optimizations

rdeits, 2016-10-28:
In mixed-integer programming, an indicator variable is a binary variable which is used to determine if a particular constraint should be active. The most common form is something like:

```
(z == 1) implies Ax <= b
```

where `z` is a binary decision variable and `x` is another (continuous, binary, or integer) decision variable. We'll call that an "indicator constraint" since it uses `z` as an indicator variable. 

We have typically transformed this type of constraint into a linear constraint with a "big-M" formulation, which looks like this:

```
Ax <= b + M * (z - 1)
```

where M is a large constant value. But choosing the correct value of M can be tricky if we want to avoid badly scaling our problem. 

However, the new version of gurobi (7.0) supports adding the indicator constraint directly to the gurobi model, with a syntax like:

```
model.addGenConstrIndicator(z, 1, A * x, LESS_EQUAL, b)
```

The gurobi manual describes the way they handle these constraints on pg. 515: http://www.gurobi.com/documentation/7.0/refman.pdf

I'm opening this issue to start a discussion on whether these new constraint types are something we want to support directly in MathematicalProgram.

rdeits, 2016-10-28:
As a first experiment, I tried creating a simple region-assignment problem (essentially a very basic footstep planner). The optimization tries to find x, y, and z near some goal, with the restriction that x, y, and z lie within one of a fixed number of convex planar regions. I then constructed that model in both the big-M formulation and the indicator variable formulation, and got the following runtimes (each row is an average over 50 randomly generated models):

```
num_regions: 5 	bigM runtime: 0.0053172826767 	indicator runtime: 0.00305594444275
num_regions: 10 	bigM runtime: 0.0179399204254 	indicator runtime: 0.00636961460114
num_regions: 20 	bigM runtime: 0.0250561571121 	indicator runtime: 0.0202241277695
num_regions: 40 	bigM runtime: 0.0419656610489 	indicator runtime: 0.0713598155975
num_regions: 80 	bigM runtime: 0.0776280117035 	indicator runtime: 0.272704000473
num_regions: 160 	bigM runtime: 0.281060013771 	indicator runtime: 0.919965324402
```

It looks like the indicator formulation of this problem is faster for small models, but the solver time grows quadratically with the number of regions, while the bigM formulation grows roughly linearly with the number of regions. I have no idea why that would be the case.

RussTedrake, 2016-10-29:
could it be using the other trick :  x = sum_i x_if_boolean_i, yada?

rdeits, 2016-10-29:
It's possible. Unsurprisingly, the gurobi manual doesn't go into much detail on exactly how they formulate these models.

hongkai-dai, 2019-03-26:
Close it for now. Will reopen it if needed.

closed 2019-03-26 by hongkai-dai