RobotLocomotion/drake #3951: Error when installing Gurobi

sylviadai, 2016-10-29:
Hi! I am installing drake from source and I followed the guide in issue #184 when installing Gurobi. After I finished the steps git clone and run make in gurobi, and then when I cd into drake and try to run make configure, I met the following error:

CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR
  GTEST_MAIN_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindGTest.cmake:205 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /home/sylvia/Desktop/drake-distro/cmake/packages.cmake:50 (find_package)
  CMakeLists.txt:138 (drake_find_package)

My OS is Ubuntu 16.04 and I tried both Matlab 2016b and Matlab 2014b, but I always run into this error. Can you help me find out what problem it is and tell me how to solve it? Thanks very much!

naveenoid, 2016-10-29:
@sylviadai ..the method for installing gurobi is described here : http://drake.mit.edu/from_source.html#using-gurobi-as-an-external
The method described in #184 is no longer valid.

liangfok, 2016-10-30:
I'm going to close this issue assuming the link to the latest instructions solved the problem. Please feel free to re-open if the new instructions still don't work.

closed 2016-10-30 by liangfok

pvarin, 2016-11-21:
The latest instructions are kind of ambiguous for Mac users. The Gurobi site (https://git.io/vXVsE) suggests that Mac users download the .pkg file, but the error message in the build system says
```Please download the Gurobi 6.0.5 archive (.tar.gz) and set ...```

I've tried both downloading the .pkg file and the .tar.gz file and I continue to get errors.

david-german-tri, 2016-11-21:
@pvarin Yep, those directions were only tested on Linux.  When you figure it out on Mac, please submit a PR to update the documentation!  (Also, @jamiesnape and @BetsyMcPhail may have some hints, since this works on CI somehow...)

jamiesnape, 2016-11-21:
`export GUROBI_DISTRO=/path/to/gurobi6.0.5a_mac64.pkg`

Jerryxiaoyu, 2018-11-01:
I was trying to re-implement the old matlab version drake. But I can't find `gurobi6.0.5_linux64.tar.gz`, because this version is too old to be deleted in  [](https://www.gurobi.com/downloads/download-center)
Could anyone send me a copy? Thanks.
@jamiesnape @pvarin

jamiesnape, 2018-11-01:
https://packages.gurobi.com/6.0/gurobi6.0.5_linux64.tar.gz

Jerryxiaoyu, 2018-11-01:
@jamiesnape Thanks a lot. It is very helpful.