RobotLocomotion/drake #3952: simulator needs a real-time mode

RussTedrake, 2016-10-30:
I’ve heard it discussed, but wanted to make a ticket out of it.  We need support for having the integrator synchronize it’s clock with the system clock.  

it should be easy.  the system 1.0 implementation is here:
https://github.com/RobotLocomotion/drake/blob/7b8abb4d670a1173336d73e9112e3e257dcffb17/drake/system1/Simulation.h#L44

liangfok, 2016-10-30:
FTR, here are two previous discussions on adding real-time factor support to Simulator 2.0:

https://reviewable.io/reviews/robotlocomotion/drake/3202#-KQ5QuavVi_hLXuHjKrv

https://reviewable.io/reviews/robotlocomotion/drake/3202#-KQ5StG9UV2NsbVT6yA6

RussTedrake, 2016-10-30:
thanks @liangfok .  @sherm1 -- given your feedback in those posts, let me clarify.  my ask is for slowing down the sim to real time speeds (not about emphasizing performance).  debugging even simple systems, like the pendulum, is very difficult when the playback in the visualizer is completely divorced from physical reality.  i personally get a lot of intuition from the physical real-time playback speed.

edrumwri, 2016-10-30:
Russ, if you're intending this for visual inspection/debugging, why
would this slowdown functionality not belong in the visualizer? I
think this is the only place where it will be used. I anticipate that
the visualizer would listen for messages at the appropriate time and
then visualize them when they arrive, even if the simulator is now
working on simulating far into the future.

In the past, I've thought quite a bit about the more general problem:
how do you slow "client" software (which could be a visualizer or even
a motion planner) such that it progresses against a simulator's
virtual time as it would against real time hardware (like robots
running in the real world)? See
http://robotics.gwu.edu/positronics/wp-content/uploads/2014/11/taylor2014_2.pdf

The "real time factor" is also a kludge: it's primarily useful for
visualization and only then if the simulation runs at a constant rate
and faster than real-time.

Regards,
E

RussTedrake, 2016-10-30:
I think it's more general than just for visualization.  For example, we may have controllers with dynamics that must integrate in real time to be compatible with real hardware.  Some other tools that we might want to interface with only know how to run at real-time rates.

sherm1, 2016-10-30:
@edrumwri and I had been tossing around the idea of creating a "SlowDownSystem" that could be added to a system Diagram. Its Publish() method would get called frequently and would delay until sim time matches real time. That would require no support from the Simulator. 

At the moment I'm inclined just to add a slowdown option to Simulator, but I thought the slowdown System idea was kind of cool. Any thoughts?

RussTedrake, 2016-10-30:
The slowdown system is actually exactly how i did it in matlab.  (But we called it a realtime block).  Either way works for me.

The logic in the system1 code is worth looking at.  Little details like warning (not erroring out) if we get behind realtime, and warning only once, took a few iterations to get right.

liangfok, 2016-10-30:
Here is the system1 code: https://github.com/RobotLocomotion/drake/blob/528991b15b5486c5b0f4e1d1697a3ed29997bb50/drake/system1/Simulation.h#L100

amcastro-tri, 2016-10-31:
I like the idea of a system as you suggest @sherm1, I also think it's a cool one. I vote up for `RealTimeBlock` or `RealTimeSystem` as @RussTedrake suggested.

jwnimmer-tri, 2016-10-31:
If the resolution is to offer a `SlowDownSystem`, be sure that the simulator promises that `SlowDownSystem::Publish` will be invoked even when the system has no inputs and no outputs.  (In other words, it is an anti-requirement that the `SlowDownSystem` have inputs.)

It's also worth considering what happens when someone adds more than one `SlowDownSystem` to a `Diagram`.  I guess "don't do that", perhaps?

sherm1, 2016-10-31:
Not that I would recommend it, but I don't think having multiple SlowDownSystems would be a problem. The first one's Publish() method would suck up all the extra time. The second one's would find the time even with real time so would return immediately. (Give or take a few microseconds.)

Calls to Publish() are uncorrelated with input/output ports. A SlowDownSystem wouldn't need either.

jwnimmer-tri, 2016-10-31:
If `SlowDownSystem` has a configurable time scale factor (as System 1.0 did, and which will be a commonly-requested feature), then having multiple instances with different factors wouldn't be as trivial.

edrumwri, 2016-10-31:
A few more questions: do we actually want a "real time factor" (a la
Gazebo) or simply a mode where the simulation throttles itself to running
in real-time? And what are the accuracy requirements for the throttling?
I've had enough experience with timing on multi-user, multi-core,
non-realtime systems to know that different accuracy requirements entail
different solution approaches.

sherm1, 2016-10-31:
> having multiple instances with different factors wouldn't be as trivial

Not *as* trivial, but pretty close -- the slowest one would always win.

sherm1, 2016-11-17:
Although the SlowDownSystem is kind of cool, I think we should do this by adding  a "real time governor" option to the Simulator.

RussTedrake, 2016-11-21:
fwiw -- am now blocked on this for the acrobot work.

sherm1, 2016-11-21:
on my list for this week

sherm1, 2016-11-22:
@edrumwri is concerned that this feature could be confused with an *actual* realtime simulation, which can't really be achieved with a non-realtime OS like Ubuntu or OSX. So far I've only seen it used for visualization, but from @RussTedrake's comment above I wonder if it will be used in other ways. If so, is the fuzzy-realtime capability provided here adequate for those purposes?

RussTedrake, 2016-11-22:
I'm not particularly worried about that, but it's a fair point.  Perhaps a different name, like "system_clock_factor" (i don't particularly like that one) could be better?  I'm ok either way.

sherm1, 2016-11-22:
Left the name as `Simulator::set_target_realtime_rate()` but added weaselly warning comments in #4209.

PR #4209 (2016-11-22): Add realtime slowdown feature to Simulator
Adds `Simulator::set_target_realtime_rate()` to slow down a too-fast simulation. Let's hope we have this problem a lot!

Fixes #3952.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4209)
closed 2016-11-22 by sherm1