RobotLocomotion/drake #3956: AffineSystem should implement LeafSystem<T> for multiple scalar types

RussTedrake, 2016-10-31:
proposal:
I took a look at replacing
``
template <typename T>
class AffineSystem : public LeafSystem<T> {
``
with 
``
class AffineSystem : public LeafSystem<double>, public LeafSystem<AutoDiffXd> {
``
and got most of the way through doing it.  But there are a few small pain points* that I worry are symptomatic of a larger resistance to this approach, so I wanted to make sure we discuss before I hammer them out.

Motivation: dramatically simplify the analysis of more complex systems.  For example, I'm writing the `TimeInvariantLqr` method for System2.0.  This will return an `AffineSystem`, which can then be put in feedback combination with the original system.  If I wanted to analyze the closed-loop system via `AutoDiff` as well as simulate it, I would currently have to call `TimeInvariantLqr` twice to make two different `AffineSystems` (templated on different scalar types).  If I wanted to do polynomial analysis of the closed-loop system -- and I do! -- then I have to construct the `TimeInvariantLqr` with a `Context<double>` but ask it to return an `AffineSystem<Polynomial>`.  That's starting to get more ugly.  

I fear that the initial promise of having System2 systems implement multiple scalar types might be slipping away.  @david-german-tri , @sherm1 -- hoping for your thoughts on this.

`*` making an `template<T> EvalOutputsImpl()` etc was simple enough, but I didn't like having to call `System<double>::DeclareInputPort` and `System<AutoDiffXd>::DeclareInputPort` in the constructor in order to get everything initialized properly.  This one is clearly fixable, but do the problems run deeper?

sherm1, 2016-10-31:
@RussTedrake, I favor the approach I prototyped in #2840 which I think is much cleaner than multiple inheritance (which I can't make work all the way down through RBTree). The idea is to construct the entire System identically once per instantiated scalar type. All that we are missing for this to work now is automated conversion from `Context<double>` to `Context<T>` for other scalar types T. As I understand it, that is very similar to the current conversions that are done from `KinematicsCache<double>` to `KinematicsCache<AutoDiff>`. 

Creating and managing the multiple objects can be automated too. But I think we could get the method working and useful before doing that.

RussTedrake, 2016-10-31:
Great.  I remembered that proposal, but didn't know how specific it was to MBTree.  Is your thinking that it would be implemented (repeatedly) for every system of interest... like AffineSystem?  or that we would somehow be able to capture this at the System or LeafSystem level?

sherm1, 2016-10-31:
My intention is that support would be provided at the System/Context level. For the "manual" version, user code has to invoke the identical Diagram constructor once for each instantiation and keep the resulting Diagrams available. That doesn't require any System support, but we still need to support initialization of a `Context<T>` from a `Context<double>`. For the automated version we have to be able to construct a `System<T>` given a `System<double>`. That's basically a copy constructor except for the mismatched scalar type. That requires that any contained classes are able to perform that same copy construction recursively. With that in place the user sees only a single construction of the `System<double>` object and can't cause mismatched instantiations.

RussTedrake, 2016-10-31:
this might actually be the right/shortest path for me to get lqr working with system2.  Would like to discuss it f2f today if possible

jwnimmer-tri, 2016-10-31:
Ah, the transmogrify discussion has started up, before I got a chance to finish my draft on sparsity.  If I happen to be around for the discussion, I would be happy to participate.  I think the concept in #2840 (ask a system to transmogrify itself) is broadly correct.  The next details are (1) System member fields (of both non-Scalar, Scalar, and Scalar-dependent types) are the biggest challenge, be sure to hone in on those in a few experiments, (2) decide whether the type domain of the transmogrify relation open or closed, and (3) when the Scalar supports "Variable" (e.g. FunctionalForm or SymbolicExpression), it's not obvious how you can ask a Context<double> to transmogrify itself into Context<FunctionalForm> without writing a FunctionalForm-specific specialization; is that we best we can do?

david-german-tri, 2016-10-31:
At a high level, :+1: for the #2840 approach and :-1: for multiple inheritance.  Here are a couple of additional thoughts:

> (1) System member fields (of both non-Scalar, Scalar, and Scalar-dependent types) are the biggest challenge, be sure to hone in on those in a few experiments,

I think we can simplify further by dropping the goal of "alternates" and System transmogrification.  Specifically, I don't think this automation is useful as a framework feature:

> For the automated version we have to be able to construct a `System<T>` given a `System<double>`. That's basically a copy constructor except for the mismatched scalar type. That requires that any contained classes are able to perform that same copy construction recursively.

 From a framework perspective, a `System` is const after construction.  There are literally no const public methods on `System<T>` other than the `set_parent` shrapnel.  For specific Systems that don't add mutators, you can get a `System<AutoDiff>` identical to your `System<double>` just by calling the constructor with the same arguments.  For Systems that do mutate, I think it's going to be highly case-by-case whether you would rather have a factory method or a recursive copy, and there's not much the framework can abstract away for you in either case.  I say leave this part of the problem to System authors.

----

That leaves `Context` transmogrification:

* For numerical fields and numerical types (e.g. double to AutoDiff), trivial.
* For numerical fields stuffed inside an AbstractValue, the best I can come up with is to add a transmogrifier function to AbstractValue, which seems possible but kind of wonky; I propose we defer that until there's a need.
* For numerical to non-numerical types, my knee-jerk reaction is that transmogrification is actually meaningless and shouldn't be supported.  Just make a new Context.  But @RussTedrake's initial post suggests I may be wrong about that...

david-german-tri, 2016-10-31:
>  I say leave this part of the problem to System authors.

Well, let me walk that back partially.  We might want to transmogrify a Diagram automatically by transmogrifying all of its constituent Systems, which perhaps was @sherm1's original point?  I have some implementation-detail questions about that which I will take to #2840, but in any case I agree with Sherm that it's not blocking for Russ's request.

david-german-tri, 2016-10-31:
@jwnimmer-tri points out that framework-level System transmogrification is useful at least to FunctionalForm/SymbolicExpression for the default implementation of sparsity.

I'm increasingly in favor of framework-level transmogrification on a closed set of types, and system-level transmogrification on any custom types.

david-german-tri, 2016-10-31:
Regardless of all of the above, here's a thing that currently has me stumped: when upconverting a Context from `double` to `AutoDiffScalar`, how do we initialize the partials?

sherm1, 2016-10-31:
> when upconverting a Context from double to AutoDiffScalar [ad], how do we initialize the partials

Per f2f: Zeroes. Then caller uses `system_ad.usual_methods(context_ad, args_ad)` to set the autodiff weightings on the inputs of interest in `context_ad`, then calls `system_ad.eval_whatever(context_ad)`, then looks at the derivative part of the outputs.

david-german-tri, 2016-10-31:
> Per f2f: Zeroes.

Specifically, a partials vector of size zero; the caller/user-code needs to set the number of partials as well as their weighting, in accordance with whatever semantics the caller needs.

david-german-tri, 2016-11-22:
Tentatively closing this as superseded by #2840 and #3967.

closed 2016-11-22 by david-german-tri