RobotLocomotion/drake #3973: Remove drake_export.h

jwnimmer-tri, 2016-11-01:
Per #3965, we need to bake some perl pie and remove the include file itself, as well as the `DRAKE_`EXPORT annotations.

jwnimmer-tri, 2016-11-05:
Ongoing use is causing compiler warnings (marking a symbol public when it already is).  I'm going to at least remove the use of the macro.  Then we can delete the file in a second step (to allow for easier downstream transitions).

jwnimmer-tri, 2016-11-20:
Dropping priority; the only remaining task is to `rm` the header file at some unspecified future date.  (For example, once feature branches that depend on it get updated to account for the new deprecation warning that will show up.)

jwnimmer-tri, 2016-11-30:
Apparently my #4292 magic phrase didn't auto-close this, but this issue is resolved now.

closed 2016-11-30 by jwnimmer-tri

liangfok, 2016-11-30:
Looks like the magic phrase only works when it's in the OP.