RobotLocomotion/drake #3977: Purge System 1.0 from kuka_iiwa_arm

jwnimmer-tri, 2016-11-01:
Subtask of #3960. Purge System 1.0 code from examples/kuka_iiwa_arm.

jwnimmer-tri, 2016-11-01:
This involves several subtasks related to System 2.0 support of gravity compensation, etc, etc, but I believe that nothing else in #3960 porting will conflict or duplicate with that work, so I think this issue is an independent task from the others.

naveenoid, 2016-11-24:
I think the following can be removed without any problems in the first phase: 
run_kuka_iiwa_gravity_compensated_torque_control.cc
run_kuka_iiwa_gravity_compensated_position_control.cc
test/kuka_iiwa_dynamics_test.cc
test/kuka_iiwa_gravity_compensated_position_control_test.cc
test/kuka_kuka_iiwa_gravity_compensated_torque_control_test.cc
test/kuka_iiwa_pd_control_test.cc
Im addressing it in the next PR.

jwnimmer-tri, 2016-11-30:
It looks like only `kuka_lcm_visualizer.cc` remains to be purged (at which point `kuka_simulation` can go away also), is that correct?

sammy-tri, 2016-11-30:
I don't think kuka_simulation is planned for removal...  it's System2.

jwnimmer-tri, 2016-11-30:
Sorry, I meant `iiwa_simulation`.

jwnimmer-tri, 2016-12-06:
Is there anything I can do to help `kuka_lcm_visualizer` go away?  Is it just a matter of rm'ing in a PR, or is it under active use?

sammy-tri, 2016-12-07:
I suppose rm'ing it would be one way of forcing me to recreate it differently the next time I need it.

More seriously, I think the need for it is reduced now that kuka_iiwa_arm has it's own director app and robot state translator.  I think I'd approve of a PR deleting it.

PR #4410 (2016-12-07): Remove System1 code from kuka_iiwa_arm
Closes #3977.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4410)
closed 2016-12-07 by jwnimmer-tri