RobotLocomotion/drake #3979: Purge System 1.0 from ROS integration

jwnimmer-tri, 2016-11-01:
Subtask of #3960. Purge System 1.0 code from drake-distro/ros.

This is mostly independent of #3978, since this code really only depends on RBS.

To be clear: deleting code that depends on System 1.0 is a valid solution.

jwnimmer-tri, 2016-12-06:
@liangfok It seems like with the HSR ROS demos coming online, that's a good example of System 2.0 use within ROS.  What's a smart plan for removing System 1.0 support from ROS demos?  Do we need to port any more code, or can we just remove anything that depends on System 1.0?

liangfok, 2016-12-06:
I will purge all remaining System 1.0 code and rely on Git history for reference as I port the System 1.0 code to System 2.0. A PR for this will be coming soon.

PR #4391 (2016-12-06): Removes System 1.0 code from drake-distro/ros.
Resolves #3979.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4391)
closed 2016-12-06 by david-german-tri