RobotLocomotion/drake #3984: drake_add_example() Should Be Generalized to Allow Custom Destination Directory

liangfok, 2016-11-01:
In `drake-distro/cmake/examples.cmake`, I would like to generalize function `drake_add_example()` to allow the user to specify the directory in which the example's submodule is placed. It currently places all submodules in `drake-distro/drake/examples/`. For some examples that are actually ROS packages, I would like to place them in `drake-distro/ros/`. Should a new `drake_add_ros_example()` method be created? Or, should the existing method be modified to accept another parameter?

liangfok, 2016-11-01:
@mwoehlke-kitware I assigned this issue to you since I believe you were the original author of `drake_add_example()`. Please feel free to reassign.

jwnimmer-tri, 2016-11-01:
@liangfok What are you _actually_ trying to do?

liangfok, 2016-11-01:
There is a private repository containing a ROS package that I need to add to Drake. I'd like Drake's build system to pull it in when the user executes something like the the following:

```
$ cd drake-distro/build
$ cmake .. -DEXAMPLES_FOO=ON
$ make -j 8
```

Since the repository is private, it needs to be by default `OFF`.

liangfok, 2016-11-01:
Thinking about this some more, I think I'd like to have a `drake_add_ros_package()` function that adds a submodule to `drake-distro/ros/`. It's less general than allowing the user to specify an arbitrary location, but I don't need the extra flexibility at this moment.

mwoehlke-kitware, 2016-11-01:
I am trying to understand... why is this "like an example" (or even, "neither like an example *nor* like other externals") as opposed to "like other externals"? I could offer a more meaningful opinion if I knew...

liangfok, 2016-11-01:
Sorry for being confusing. Let me think about this some more. I'll assign this issue to myself for now.

liangfok, 2017-02-16:
I'm going to close this issue because I believe `drake_add_external()` is sufficient for my needs. See, for example, this:

https://github.com/RobotLocomotion/drake/blob/09bf14b6fd82167f19e6c0845957e8664e064633/CMakeLists.txt#L365

closed 2017-02-16 by liangfok