RobotLocomotion/drake #3994: Quadrotors need porting to System2.0

naveenoid, 2016-11-02:
# Problem Definition
The current Quadrotor example needs porting over to System2.0. Building in this functionality will also be a useful exercise to drive the state-of-the-art in several aspects of Drake's simulation ability as well as TrajectoryOptimization.  This task will be organised in a set of spirals similar to PR https://github.com/RobotLocomotion/drake/issues/3919 

# Spirals
- [ ] Spiral 1
- [ ] Spiral 2
- [ ] Spiral 3

# Spiral Details
## Spiral 1 - Quadrotor given unity constant vector input intialized in a "world"
A direct port of runDynamics.cpp

## Spiral 2 - Quadrotor initialised from a set of initial positions moves to a target position and hovers
A direct port of runLQR.cpp 

## Spiral 3 - Trajectory optimization on the Quadrotor
Possibly to navigate the world setup in Spiral1.

RussTedrake, 2016-11-02:
duplicate of #3975.  and not a high priority for the manipulation team.

naveenoid, 2016-11-02:
@RussTedrake This issue was just a placeholder for a student of mine at ETH. I self-assigned since he is not yet a contributor. I guess we can follow the progress from #3975.

RussTedrake, 2016-11-02:
sorry — i realized as I hit the button that my text sounded short.  :)  you can reopen and the duplicate is noted.

tkoolen, 2016-11-03:
Would it make sense to just use a URDF + RigidBodyPlant instead of reimplementing the dynamics for this special case?

naveenoid, 2016-11-03:
Thats exactly the plan. Will greatly simplify future work on this example.

RussTedrake, 2016-11-03:
I consider the special case dynamics to be a useful unit test to ensure the dynamics (including force/torque input) are as expected.  Perhaps they can continue to exist in that capacity (in the same spirit as the pendulum test which checks the urdf dynamics vs the analytical dynamics)

naveenoid, 2016-11-08:
Progress on Spiral 2 and 3 are hinged on resolving https://github.com/RobotLocomotion/drake/issues/4076

jwnimmer-tri, 2016-12-08:
How inconvenient would it be to have the System 1.0 version removed from the source tree before the System 2.0 version is added?  (I'm thinking of e.g. running the old version while developing the new one.)

This issue is about adding the new feature, but we could still remove the old code and close out #3975, if that didn't disrupt this work too much.

RussTedrake, 2016-12-08:
I strongly support this.  I think we should zap the system1 folder now.  The atlas example is also using it, but can be removed too.

naveenoid, 2016-12-08:
Zapping system1 wont affect my groups work. Lets go ahead..I can review your zapping PR if needed. (Tagging @foehnph)

RussTedrake, 2016-12-08:
also tagging @psiorx .  He is the author, and still a user of the system1 quadrotor sim.  But I think not on master.  And @psiorx -- it's an easy port to system2 at this point.

jwnimmer-tri, 2017-06-19:
See also #6243.

RussTedrake, 2018-11-07:
I think this one can be safely closed.  (Certainly it's current phrasing is far out of date).  @naveenoid -- please reopen if I'm wrong.

closed 2018-11-07 by RussTedrake