RobotLocomotion/drake #4000: valkyrie_balancing_system_test does not compile

jamiesnape, 2016-11-02:
Two bugs in play here. One is fixed by #3996, but it exposes a second:
```
error: object of type 'drake::examples::qp_inverse_dynamics::HumanoidStatus' cannot be assigned because its copy assignment operator is implicitly deleted
```

https://drake-cdash.csail.mit.edu/viewBuildError.php?onlydeltap&buildid=171160

david-german-tri, 2016-11-02:
Why doesn't this show up in CI?

jamiesnape, 2016-11-02:
#3996

jamiesnape, 2016-11-02:
snopt_c_FOUND was not defined. That test has never run.

david-german-tri, 2016-11-02:
@jamiesnape is disabling both `valkyrie_balancing_test` and `valkyrie_balancing_system_test` in #3996 for now.

siyuanfeng-tri, 2016-11-04:
I believe they are fixed in https://github.com/RobotLocomotion/drake/pull/3907, waiting for review to finish.. 
I apologize for the delay.

closed 2016-11-06 by siyuanfeng-tri