RobotLocomotion/drake #4005: AutoDiffScalar needs isnan() and isinf() overloads

edrumwri, 2016-11-02:
This is necessary for adjusting step sizes in the context of variable step integration.

david-german-tri, 2016-11-02:
Seems reasonable.  It should be pretty trivial to add this in https://github.com/RobotLocomotion/drake/blob/master/drake/math/autodiff_overloads.h mimicking the existing code.  Any particular reason for assigning it to me?

edrumwri, 2016-11-02:
I was told that you had written the other code in autodiff_overloads.h, so I figured you "owned" the file. I'm happy enough to write it myself if you prefer.

david-german-tri, 2016-11-02:
> I was told that you had written the other code in autodiff_overloads.h, so I figured you "owned" the file. I'm happy enough to write it myself if you prefer.

Some of it came from RLG, some from me.  If you need it on a particular timetable, probably best that you write it yourself.

edrumwri, 2016-11-03:
Addressed by pull request #4008

closed 2016-11-03 by edrumwri