RobotLocomotion/drake #4017: Option to abort simulator if any state is inf or nan

david-german-tri, 2016-11-03:
Forwarding from @jwnimmer-tri in automotive weekly meeting: it would be nice if the simulator had an option that caused it to inspect the state on every iteration, and abort if `inf` or `nan` appears anywhere.

sherm1, 2016-11-03:
I agree this would be a good option to have. Not a good default though, at least for continuous systems.

In continuous systems the state gets NaN'ed because the state derivative went bad. In a variable-step context the default response is to assume that the trial step was too big and pushed the underlying computation out of its defined domain. (This is one of the many ways an arbitrary derivative function can fail.) The usual result is that the integrator cuts the step size, which typically avoids the out-of-domain problem altogether as the trajectory evolves. If the step size reaches its minimum allowed value but still produces bad derivatives, the simulation terminates reporting the error.

In the default case we should keep stats on how many times we had to reduce the step size due to a failed derivative call; that can often be helpful in debugging poor performance.

Adding @edrumwri.

sherm1, 2016-11-03:
Here's an example: I decide to use roll-pitch-yaw angles to parameterize a system that I know will never get anywhere near the singularity caused by pitch=pi/2. The integrator decides to take an overly-ambitious trial step that results in a trial state in which pitch is very close to pi/2. The derivative can't be computed correctly there, and the author had already determined that the trajectory would never get there, so the code is not hardened for that case. That's going to result in an error or a NaN, but is completely harmless and easily resolved by reducing the step size since the correct trajectory doesn't go there.

jwnimmer-tri, 2016-11-03:
I think only rejecting "committed" state updates that are Inf / NaN would be good enough.  It's really just meant to catch simple PEBKACs in developing new code, where something in EvalTimeDerivatives or EvalOutput divides by zero accidentally or whatever, and instead of failing-fast right now the simulation runs on and you're like "WTF why is my car doing that" only to see your math typo and NaN during much debugging days later.

/CC @maddog-tri

jwnimmer-tri, 2019-09-19:
I have lived OK without this for 3 years, so I guess it wasn't a big deal.  Request withdrawn.

closed 2019-09-19 by jwnimmer-tri