RobotLocomotion/drake #4018: Clarify distinction between spatial forces and wrenches

sherm1, 2016-11-03:
A terminology finer point:
- a *spatial force* consists of two 3-vectors, a torque and a translational force
- a *wrench* is a specific kind of spatial force in which the torque component is an `r X f` moment of the force component

Every spatial force requires a contextually-specified frame in which the two 3-vectors are understood to be expressed, and at whose origin the translational force is considered to be applied.

The need for this clarification came up in review of the minimum moment point (aka center of pressure) PR #3985. There, a collection of contact wrenches is accumulated into a contact resultant, which is a *spatial vector* but not necessarily a *wrench*. We will attempt to be precise about this distinction but use good comments to eliminate ambiguity. We'll use a `SpatialForce` typedef when we don't actually have a wrench. And so should you!

mitiguy, 2016-11-07:
There are many qualiﬁers for forces which have meaning in a particular context, e.g., contact and distance forces, internal and external forces, workless and contributing/non-contributing forces, applied and constraint forces, normal and tangential forces, spring/damper/gravity force, etc.  

I always use the qualifier "spatial" with "spatial force".

My preferred vocabulary:

"Real force" -- has two points and a vector directed parallel to line connecting those two points, i.e., follows Newton's 3rd law.

"Force"  --  associated with at least one point of application, e.g., force on point p.

"Resultant force"  -- not necessarily associated with a point of application.

"Moment of a force about point O" -- Cross product of a position vector (e.g., from point O to point P) with a force on point P.  

"Torque"  -- Moment of a set of forces whose resultant is zero.  Is usually associated with a rigid object. 
Unlike a generic "Moment of a force or set of forces" (which are associated with a "moment center" O, 
torque is the same, regardless of the choice of moment center.

siyuanfeng-tri, 2016-11-28:
siyuanfeng-tri
Just so that I understand clearly, when you say wrench, you mean [r x f, f], and it doesn't contain a "pure" torque part. The torque is coming from reference point being different from the force is applied?
And we are using spatial force for everything else, which is generally concatenation of some torque and force?

---------------------------------------------------------------------------------------
 SeanCurtis-TRISean Curtis
Correct. It's also defined in the comments in eigen_types.h (https://github.com/RobotLocomotion/drake/blob/master/drake/common/eigen_types.h#L104)
1 HOUR AGO

---------------------------------------------------------------------------------------
siyuanfeng-tri
yeah, I saw these, but just want to be 100% clear.
14 MINUTES AGO

---------------------------------------------------------------------------------------
 tkoolenTwan Koolen
wrench is just force + torque part
Correct.

For example, @tkoolen provided a reference where the wrench is defined to be the [ r x f, f] construct
No, that's a misinterpretation of that definition (https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjz3869xczQAhXCsFQKHYPIBLwQFggdMAA&url=http%3A%2F%2Fbibtexbib.free.fr%2Fbibliographie_net%2Fduindam_phdthesis.pdf&usg=AFQjCNH7KIsyN5AD-OfcDegTZMonqXUyKQ&sig2=p4_MS1Up7yI6ksrvnam3jA, definition 2.13). The wrench acting on a body is just a 6-vector consisting of a torque part (rotational component) and a force part (linear component). The way this wrench relates to the individual forces acting on the body, and how it is defined in terms of those forces, is wrench = [torque; force] = [sum_i r_i x f_i; sum_i f_i]. But it would not make sense to store the individual application points and the individual forces (r_i's and f_i's) in a wrench class.

I strongly disagree with this particular distinction between spatial force and wrench, and I haven't seen any authors make this particular distinction. Featherstone makes a bit of a vague distinction between wrenches and spatial forces based on screw representation, but in the end says that elements of the motion space M^6 and the force space F^6 can be regarded as twists and wrenches, respectively. Murray uses 'wrench' exactly in the way Featherstone uses 'spatial force', except that Featherstone also refers to spatial momenta as spatial forces (which I find unfortunate).
11 MINUTES AGO

I am copying some discussion in #4238 

---------------------------------------------------------------------------------------
siyuanfeng-tri
I agree with the first part of what Twan said, and I also found it weird to make THE particular distinction between wrench and spatial forces. I don't clearly see why that's necessary.
For my simple 8bit brain, all I get is people are unhappy with me using wrench because they think it really means something very specific, and spatial force contains all I want to say. So I am just pattern matching wrench to spatial force.
The reason I prefer wrench is because that's what I have been taught with, and more importantly it's shorter to type.

SeanCurtis-TRI, 2016-11-28:
I think there may be a misunderstanding here.  Twan made a comment: 

  > But it would not make sense to store the individual application points and the individual forces (r_i's and f_i's) in a wrench class.

No one is advocating this.  The difference between the `WrenchVector` and `SpatialForce` is that the `WrenchVector` guarantees that the relationship between the torque and the force is `r X f`.  The point `r` is now gone, and it's only meaningful w.r.t. to a particular origin (or as definition 2.13 defines it the origin of coordinate frame \Sai_i.)  By this convention, if a function returns a `WrenchVector`, this relationship should be guaranteed.  It excludes the possibility of pure torques being rolled in.

The `SpatialForce` is the catch all where the torque term can include pure torques -- torques that are independent of the translational force.

This distinction came up because, in earlier PRs, we used the term "wrench" for a variable and that immediately gave rise to the question: What's the origin around which this wrench is defined?  A natural enough question for the wrench as defined in 2.13.  But *not* a meaningful question w.r.t. an arbitrary spatial force.

tkoolen, 2016-11-29:
> No one is advocating this.

OK, great!

> The difference between the WrenchVector and SpatialForce is that the WrenchVector guarantees that the relationship between the torque and the force is r X f.

and

> The point r is now gone, and it's only meaningful w.r.t. to a particular origin (or as definition 2.13 defines it the origin of coordinate frame \Sai_i.) [...] [A WrenchVector] excludes the possibility of pure torques being rolled in.

For me, r x f is the very definition of torque, so the distinction disappears. A pure torque can *only* arise when there are two or more forces acting at different points in such a way that the sum of the f_i's is zero, but the sum of the (r_i x f_i)'s is nonzero. This is what I meant with https://github.com/RobotLocomotion/drake/pull/3821#issuecomment-254897180 and https://github.com/RobotLocomotion/drake/pull/3821#issuecomment-254915452. The vector [tau; f] with tau = [1; 2; 3] and f = [0; 0; 0] is a perfectly valid wrench according to definition 2.13.

I guess I don't see what the concrete implementation differences between a `WrenchVector` class and a `SpatialForce` class would be (i.e. differences between their methods). Could you elaborate on this?

siyuanfeng-tri, 2016-11-29:
I guess I don't see what the concrete implementation differences between a WrenchVector class and a SpatialForce class would be (i.e. differences between their methods). Could you elaborate on this?

I am more interested in this as well

sherm1, 2016-11-29:
> I guess I don't see what the concrete implementation differences between a WrenchVector class and a SpatialForce class would be

There are no concrete differences, just as there aren't for any other uses of 6-Vectors in general. The difference is in the interpretation. A wrench results from a system of forces applied at points, that generate moments about some other point. A spatial force is more general in that it may include arbitrary torques of unknown provenance. While one may argue whether pure torques are possible in reality, they are very real in computational dynamics because they arise from the idealized models that we use, such as joint actuators and torsional friction. When in doubt, please feel free to use SpatialForce as the more general term.

SeanCurtis-TRI, 2016-11-29:
Just to echo @sherm1.

In the physical world there are no moments without forces being applied at the end of moment arms.  In the simulated world, where we fudge things, there *are* torques without forces.  So, although we're modeling the real world, our model introduces abstractions which only exist in the model.  Our code must reflect the model.

tkoolen, 2016-11-29:
I'd just like to stress that according to definition 2.13, a wrench is not just [tau; f] = [r x f; f] (from the issue description), but **[tau; f] = [sum over i of r_i x f_i; sum over i of f_i]**. The **sum** is what makes it possible to represent a pure torque (where the linear component of the wrench is zero) as a wrench. For example, you could have f_1 = [1; 0; 0], f_2 = [-1; 0; 0], r_1 = [0; 1; 0], r_2 = [0; -1; 0], so that
tau = r1 x f1 + r2 x f2 = [0; 0; 2]
f = f1 + f2 = [0; 0; 0].
And once you just have those six numbers, it should be immaterial what exact set of forces produced the wrench (that information is lost, which is a good abstraction). So I still don't see any distinction.

mitiguy, 2016-11-30:
The word "wrench" has been used in mechanics for hundreds of years
as a special type of replacement of forces.
Although some modern roboticists use Wrench as a general force/torque,
(per the Ph.D. definition quoted), its classic meaning is more specific.
I am uncomfortable with using Wrench instead of SpatialForce.

Definition from: Chirgwin & Plumpton, 
Advanced Theoretical Mechanics: 
[In general, we can find a straight line such that an arbitrary set of
forces is equivalent to a couple whose moment is parallel to the line 
together with a force along the line.
This line is the "central axis" and this special combination 
of force and a couple is called a "wrench".]

Definition from: Kane & Levinson, Dynamics, Theory and Applications.
"A set of bound vectors consisting of a couple together with a single bound vector is called a wrench if the single bound vector is parallel to the torque of the couple."

Some background:
The classic definition of a wrench is used in conjunction with a specific way of 
replacing a set of forces by an equivalent set (e.g., useful for rigid body analyses).

A general (not necessarily a wrench) way to replace a set S of forces on a rigid body B is to:
1.  Calculate the resultant F of the set of forces (sum the forces).
2.  Choose an arbitrary point Bi of B.
3.  Calculate the moment M of the set S of forces about point Bi.
4.  Apply a force F to point Bi and apply a torque of moment M to body B.
Note: A torque is defined as the moment of a set of forces whose resultant is zero.
Note: A "couple" is defined as a set of forces whose resultant is zero.   
Hence: A torque is a special type of moment.

A special way to do replacement is with a "wrench".
A wrench is special, namely it:
1.  Minimizes the magnitude of M.
2.  Ensures M is parallel to F.

A wrench requires a special location of Bi
(Bi is said to have to lie on the "central axis" of S).

Summary:  A wrench is a special replacement.
The SpatialForce class is more general.

https://books.google.com/books?id=KnRqDQAAQBAJ&pg=PA48&lpg=PA48&dq=wrench+mechanics+minimum+magnitude&source=bl&ots=E5msWAiDSb&sig=6sWNjgOC3bXAgF2Ib27HalWuxVI&hl=en&sa=X&ved=0ahUKEwiEi7nGltHQAhVmilQKHQpCBMcQ6AEIMzAE#v=onepage&q=wrench%20mechanics%20minimum%20magnitude&f=false

https://books.google.com/books?id=-xISBQAAQBAJ&pg=PA58&lpg=PA58&dq=definition+2.13+wrench&source=bl&ots=Xc4FKjqyxN&sig=F85C5W0ZbhsuJAfuKhATihpSqgc&hl=en&sa=X&ved=0ahUKEwik_KnjiNHQAhWiwlQKHeF_BnQQ6AEIODAH#v=onepage&q=definition%202.13%20wrench&f=false

More information/examples in .pdf format will be provided.

tkoolen, 2016-11-30:
OK, that's a valid argument for having there be a distinction. The original issue description and the subsequent discussion caused some confusion on my part; now I see that's what Sherm meant.

Murray calls that representation a 'wrench along a screw' or 'wrench in screw coordinates' (so a special kind of wrench). But I understand now if you're uncomfortable with that terminology.

sherm1, 2019-11-08:
@mitiguy do you think there is any action to be taken on this issue? It's a good discussion but it doesn't seem like the problem has come up lately. Do we still need to update documentation somewhere or can we just close this?

mitiguy, 2019-11-08:
I think this conversation can be closed, particularly since Wrench is no longer in vogue in the code.  If "wrench" arises from the ashes, we can reopen the discussion again then.

closed 2019-11-08 by mitiguy