RobotLocomotion/drake #4025: RigidBodyPlant needs named (vs numbered) output port getters

jwnimmer-tri, 2016-11-04:
See https://github.com/RobotLocomotion/drake/pull/3995#issuecomment-258405893

Users should never have to write `plant->get_output_port(0)`, the `0` is a magic number.  There should instead be an accessor like `plant->get_state_vector_port()` or similar.

Perhaps even the `get_output_port(int)` visibility should be narrowed in the base class, to encourage users to call the named accessor.

sherm1, 2016-11-04:
I agree! There are several entities in System 2 that are crying out for names IMO:
- input and output ports
- subsystems
- parameters

I would like to see these with unique pathnames so that everything in a Diagram can be easily named.

jwnimmer-tri, 2016-11-04:
String names are fine, too.  We also need (and this PR is about) compile-time-checked names.

sherm1, 2016-11-04:
Oh. Sorry, I got excited and lost my mind. You just mean some handy enumszzzzz...

jwnimmer-tri, 2016-11-04:
Well, enums are better that nothing, but still suffer from death-by-integer.  I think that `rigid_body_plant->get_state_vector_port()` is superior to `rigid_body_plan->get_output_port(RigidBodyPlantPorts::state_vector)`.

amcastro-tri, 2016-11-04:
For the usage case in #3995 I think you are asking for methods in `RigidBodyPlant` that look more like `const SystemPortDescriptor<T>& get_generalized_forces_port() const` and `const SystemPortDescriptor<T>& get_state_vector_port() const`? essentially something I can pass to `DiagramBuilder::Connect(src_port, dst_port)?

jwnimmer-tri, 2016-11-04:
@amcastro-tri Yes, exactly.  And then make the `get_output_port(int)` hidden on the RBP class, if possible.  (Users would have to upcast to System in order to call if, they really needed to.)

jwnimmer-tri, 2017-01-17:
I think @siyuanfeng-tri has already fixed this?

liangfok, 2017-01-17:
I also have a further refined version in #4766.

siyuanfeng-tri, 2017-01-17:
i think we have state_output_port(), kinematics_results_output_port() and contact_results_output_port().

PR #4766 (2017-01-19): Added Named Accessors to RigidBodyPlant's Ports and Made Plant-Centric Actuator Command Input Port Only Available in Mono-Model-instance RigidBodyPlants.
This will be more robust against future `RigidBodyPlant` port number changes due to the addition of ports.

The actuator command input ports are modified as follows:

1. There is only one model instance-centric actuator command port per model instance with an actuator. (It is now possible for the plant to not have any input ports when there are no actuators.)

2. The plant-centric actuator command port is only available when there is only one [... cut]
closed 2017-01-17 by jwnimmer-tri