RobotLocomotion/drake #4034: Mysterious CI Error - could not lock config file

liangfok, 2016-11-04:
# The Error

```
15:10:33 Cloning into 'drake/doc/textbook'...
15:10:33 Submodule path 'drake/doc/textbook': checked out 'e051917c15a513e1924465060134164c85fe5141'
15:10:33 error: could not lock config file /home/ubuntu/workspace/linux-gcc-ninja-continuous-matlab-debug/.git/modules/drake/doc/textbook/config: File exists
15:10:33 Failed to register url for submodule path 'HTMLBook'
15:10:33 Failed to recurse into submodule path 'drake/doc/textbook'
```

# Example Console Logs

* https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-gcc-ninja-continuous-matlab-debug/379/consoleFull

jwnimmer-tri, 2016-11-05:
A prior example:
https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=169296
```
Cloning into 'externals/xfoil'...
error: could not lock config file /home/ubuntu/workspace/linux-gcc-ninja-experimental/.git/modules/externals/xfoil/config: File exists
```

jwnimmer-tri, 2016-11-08:
This happens up to a few times per week or more.  Let get it onto Kitware's radar.

david-german-tri, 2016-11-08:
I have yet to be inconvenienced myself, but @jamiesnape @BetsyMcPhail, PTAL?

liangfok, 2016-11-21:
@hongkai-dai experienced this here:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental-open-source-debug/804/console

jwnimmer-tri, 2016-11-22:
Again
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-continuous-matlab-debug/494/

jamiesnape, 2016-11-22:
Has anyone observed this on a non-Ninja build?

liangfok, 2016-11-29:
I just experienced this issue here:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/1170/

Lucy-tri, 2016-11-30:
Failing with snopt: linux-clang-ninja-continuous-debug-pre-drake
Cloning into 'externals/snopt'...
error: could not lock config file /home/ubuntu/workspace/linux-clang-ninja-continuous-debug/.git/modules/externals/snopt/config: File exists
(https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-clang-ninja-continuous-debug/1061/ )

jwnimmer-tri, 2016-12-01:
Again
https://drake-jenkins.csail.mit.edu/job/linux-clang-ninja-experimental-open-source/1166/

liangfok, 2016-12-05:
Again: https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/1365/console

liangfok, 2016-12-05:
Again: https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/1368/console

liangfok, 2016-12-05:
Again: https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/1372/

liangfok, 2016-12-06:
Again: https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/1431/

jamiesnape, 2016-12-06:
We have enough information now and are trying to reproduce. Post if there is a failure that is not Linux and Ninja.

mwoehlke-kitware, 2016-12-13:
RobotLocomotion/drake-ci#62 will hopefully give us some information if it happens again, and should also work around the problem by retrying the failed git command. Builds that have git errors should show as 'unstable'. (For build-cop purposes, these can be treated as "clean", but the strace output should be inspected for clues.)

SeanCurtis-TRI, 2017-01-10:
This cropped up [again](https://drake-cdash.csail.mit.edu/viewBuildError.php?onlydeltap&buildid=301801).  This time affecting `libccd`.  Per @jamiesnape's request, this is in a mac build.

liangfok, 2017-01-10:
@SeanCurtis-TRI can you provide a link to the console log or CI build page?

SeanCurtis-TRI, 2017-01-10:
Sorry, here you go: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-ninja-nightly-minimal-debug/160/

jamiesnape, 2017-01-10:
(Aside: CDash links persist longer than Jenkins links)

SeanCurtis-TRI, 2017-01-10:
I suspected that might be the case, which is why the original posting used a cdash link.

mwoehlke-kitware, 2017-01-10:
The build should not have failed outright, however. Looking closer, there are issues *other than* the "could not lock config file" problem:

```
[9/46] cd /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug && "/usr/local/Cellar/cmake/HEAD-ce71bd9/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build --output /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/CMakeFiles/submodule-sync --  /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/ci/tools/git-wrapper.bash submodule --quiet sync --
fatal: Not a git repository: /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/.git/modules/externals/ccd
fatal: Not a git repository: /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/.git/modules/externals/ccd
[10/46] cd /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug && "/usr/local/Cellar/cmake/HEAD-ce71bd9/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build --output /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/ccd-prefix/src/ccd-stamp/ccd-download --  /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/ci/tools/git-wrapper.bash submodule update --init --recursive -- externals/ccd && "/usr/local/Cellar/cmake/HEAD-ce71bd9/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build --output /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/ccd-prefix/src/ccd-stamp/ccd-download --  /usr/local/Cellar/cmake/HEAD-ce71bd9/bin/cmake -E touch /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/ccd-prefix/src/ccd-stamp/ccd-download
FAILED: ccd-prefix/src/ccd-stamp/ccd-download 
cd /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug && "/usr/local/Cellar/cmake/HEAD-ce71bd9/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build --output /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/ccd-prefix/src/ccd-stamp/ccd-download --  /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/ci/tools/git-wrapper.bash submodule update --init --recursive -- externals/ccd && "/usr/local/Cellar/cmake/HEAD-ce71bd9/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build --output /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/ccd-prefix/src/ccd-stamp/ccd-download --  /usr/local/Cellar/cmake/HEAD-ce71bd9/bin/cmake -E touch /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/build/ccd-prefix/src/ccd-stamp/ccd-download
Cloning into 'externals/ccd'...
error: could not lock config file /Users/elcapitan/workspace/mac-clang-ninja-nightly-minimal-debug/.git/modules/externals/ccd/config: File exists
fatal: Could not set 'remote.origin.url' to 'https://github.com/danfis/libccd.git'
fatal: clone of 'https://github.com/danfis/libccd.git' into submodule path 'externals/ccd' failed
Cloning into 'externals/ccd'...
Submodule path 'externals/ccd': checked out '16b9379fb6e8610566fe5e1396166daf7106f165'
```

The failure despite succeeding on retry is due to a stupid logic error in the wrapper, fixed by RobotLocomotion/drake-ci#70. What is interesting however are the errors from the `submodule sync` attempt, which makes me wonder if *all* the "could not lock config file" errors are merely symptomatic of something having gone wrong earlier...

liangfok, 2017-01-10:
Great job sleuthing!

liangfok, 2017-01-17:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental-open-source-debug/2463

liangfok, 2017-01-26:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-ninja-experimental/2824/

liangfok, 2017-02-06:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-ninja-experimental-open-source/308/

jwnimmer-tri, 2017-03-05:
Demoting high -> medium, because the git retry logic is helping, and because in BazelDrake land we won't be shelling out to git commands for submodules anymore.

liangfok, 2017-03-06:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-cpplint-experimental/1038/

liangfok, 2017-03-09:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-ninja-experimental-open-source/1609/

liangfok, 2017-03-13:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-ninja-experimental-open-source-debug/1711/

liangfok, 2017-03-22:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-ninja-experimental/1982/

liangfok, 2017-03-26:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-ninja-continuous-release/393/

liangfok, 2017-03-30:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-ninja-experimental-open-source-debug/2314/

liangfok, 2017-04-02:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-ninja-experimental/2351/

liangfok, 2017-04-15:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-ninja-experimental-minimal/2836/

liangfok, 2017-05-31:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-ninja-continuous-release/759/

liangfok, 2017-09-26:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-clang-ninja-continuous-release/1336/

PR #7134 (2017-09-29): Remove submodules and consolidate CMake options
Closes #3499, closes #3628, closes #4034, closes #4451, closes #4551.

Ultimately, the CMake externals will be removed, but it will take a little while to get the MATLAB bindings coexisting fully with the Bazel, and this removes the # 1 cause of CI flakes and saves some CI resources.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7134)
closed 2017-09-29 by jamiesnape