RobotLocomotion/drake #4045: Externals license tracker

jamiesnape, 2016-11-07:
Toward #1766.

avl: LGPL-2.0+ AND GPL-2.0+
bot_core_lcmtypes: ????
bullet: Zlib
ccd: BSD-3-Clause
director: BSD-3-Clause (externals TODO)
dreal: GPL-3.0+ (externals TODO)
eigen: BSD-3-Clause AND LGPL-2.1+ AND LGPL-3.0+ AND MPL-2.0
gflags: BSD-3-Clause
google_styleguide: BSD-3-Clause
googletest: BSD-3-Clause
gurobi: Gurobi EULA
ipopt: EPL-1.0 (third-party TODO)
iris: BSD-2-Clause AND MOSEK Non-Commercial
lcm: LGPL v2.1
libbot: LGPL-3.0+
meshconverters: BSD-3-Clause AND LGPL-2.1+ AND LGPL-3.0+ AND MPL-2.0, 
mosek: MOSEK EULA
nlopt: BSD-3-Clause AND LGPL-2.1+ AND MIT
octomap: BSD-3-Clause
robotlocomotion_lcmtypes: ????
sedumi: GPL-2.0+
signalscope: BSD-3-Clause
snopt: SNOPT EULA
spdlog: MIT
spotless: ????
swig_matlab: GPL-3.0+, Ariz. Board Regents, Univ. Chicago, Univ. Utah/Regents Univ. Calif.
swigmake: ????
xfoil: LGPL-2.0+ AND GPL-2.0+
yalmip: YALMIP Non-Commercial
yaml_cpp: MIT

jwnimmer-tri, 2016-11-07:
BTW, this is a really helpful list, and I'm very glad to see it captured.

jamiesnape, 2016-11-07:
@liangfok Could you confirm that `drake_cars_examples` is supposed to be MIT licensed?

https://github.com/RobotLocomotion/drake/blob/master/ros/drake_cars_examples/package.xml#L15

RussTedrake, 2016-11-10:
@jamiesnape -- can you mark up where the BSD-2 and Zlib are coming from in drake?
The PATH solver is only being used in the matlab examples, so can be avoided in a non-matlab distribution.

jwnimmer-tri, 2016-11-10:
I think the `drake` line in the OP also includes the licenses of https://github.com/RobotLocomotion/drake/tree/master/drake/thirdParty which explains the BSD-2 and ZLib.  Or are you saying you'd like that directory manifest itemized in the OP for clarity?

liangfok, 2016-11-10:
@jamiesnape I think `drake_cars_examples` should share the same license as Drake itself, meaning BSD. I'll submit a PR with the fix.

jamiesnape, 2016-11-10:
The alternative licensed code is mostly in thirdParty, yes. There is also BSD-3-Clause code in the cmake that comes from CMake itself.

jamiesnape, 2016-11-10:
@liangfok specify BSD-3-Clause (use the SPDX identifier). BSD is ambiguous, with anything from 0 to 4 clauses.

RussTedrake, 2016-11-10:
@jwnimmer-tri -- the specific links into thirdParty on the OP are super useful.  that's what I was hoping for.  :)

jwnimmer-tri, 2016-11-10:
Ah, which reminds me -- don't forget that Drake also has Apache-2.0 at https://github.com/RobotLocomotion/drake/blob/master/drake/common/nice_type_name.cc.

jamiesnape, 2016-11-10:
Somehow three different heuristic license checkers (and myself) missed that one. I guess the missing second paragraph threw them off.

liangfok, 2017-04-12:
Since `hsrb_description` is no longer in master, I'm removing it from the OP.

liangfok, 2017-04-12:
I'm unassigning myself from this issue since the drake/ros license is updated in #5803.

jamiesnape, 2020-07-22:
I guess this is the best place for this, but here is the list of third-party software in tree. We have been very bad at listing copyrights and it is not always obvious which files are attached to license file, so E&OE yet again:

```
Format: https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/
Upstream-Name: drake
Upstream-Contact: https://github.com/RobotLocomotion/drake/issues
Source: https://github.com/RobotLocomotion/drake
Disclaimer: This package is not part of the Debian distribution.

Files: *
Copyright: 2012 Massachusetts Institute of Technology
           2016 Toyota Research Institute
License: BSD-3-clause

Files: bindings/pydrake/multibody/test/double_pendulum.sdf
       bindings/pydrake/multibody/test/double_pendulum.sdf.LICENSE
Copyright: 2012 Open Source Robotics Foundation
           2017 Massachusetts Institute of Technology
           2017 Toyota Research Institute
License: Apache-2.0

Files: bindings/pydrake/doc/pydrake_sphinx_extension.py
       bindings/pydrake/doc/sphinx.LICENSE.txt
Copyright: 2007-2018 The Sphinx Team
           2018 Massachusetts Institute of Technology
           2018 Toyota Research Institute
License: BSD-2-clause

Files: common/autodiffxd.h
Copyright: 2009 Gael Guennebaud
           2017 Massachusetts Institute of Technology
           2017 Toyota Research Institute
License: MPL-2.0

Files: common/copyable_unique_ptr.h
Copyright: 2015 Michael Sherman
           2015 Stanford University
           2017 Massachusetts Institute of Technology
           2017 Toyota Research Institute
License: Apache-2.0

Files: common/nice_type_name.cc
Copyright: 2014 Chris Dembia
           2014 Michael Sherman
           2014 Stanford University
           2016 Massachusetts Institute of Technology
           2016 Toyota Research Institute
License: Apache-2.0

Files: debian/*
Copyright: 2020 Massachusetts Institute of Technology
           2020 Toyota Research Institute
License: BSD-3-clause

Files: manipulation/models/allegro_hand_description/meshes/*
       manipulation/models/allegro_hand_description/sdf/*
       manipulation/models/allegro_hand_description/urdf/*
Copyright: 2013 Alex Alspach
           2018 Massachusetts Institute of Technology
           2018 Toyota Research Institute
License: BSD-3-clause

Files: manipulation/models/franka_description/urdf/*.urdf
Copyright: 2017 Franka Emika GmbH
           2018 Marko Krizmancic
License: Apache-2.0

Files: manipulation/models/iiwa_description/iiwa_stack.LICENSE.txt
       manipulation/models/iiwa_description/iiwa7/*
       manipulation/models/iiwa_description/meshes/*
[... 68 lines cut ...]
           2018 Toyota Research Institute
License: BSD-3-clause

Files: third_party/net_sf_jchart2d/*
Copyright: 2001-2011 Achim Westermann
License: LGPL-2.1+

Files: tools/lint/bzlcodestyle.py
Copyright: 2006-2009 Johann C. Rocholl
           2009-2014 Florent Xicluna
           2014-2016 Ian Lee
           2017 Massachusetts Institute of Technology
           2017 Toyota Research Institute
License: Expat

Files: tools/workspace/sdformat/3bbd303.patch
Copyright: 2020 Massachusetts Institute of Technology
           2020 Open Source Robotics Foundation
           2020 Toyota Research Institute
License: Apache-2.0

[full license text removed]
```

`YEAR Massachusetts Institute of Technology` / `YEAR Toyota Research Institute` is shorthand for Drake. I do not think we need to worry about splitting that here. Everything is in Debian format rather than SPDX for now.

jamiesnape, 2021-06-22:
See inline in various `*.bazel` and `*.bzl` files and in https://github.com/RobotLocomotion/debian-drake/blob/debian/bionic/debian/copyright and https://github.com/RobotLocomotion/debian-drake/blob/debian/focal/debian/copyright.

closed 2021-06-22 by jamiesnape