RobotLocomotion/drake #4065: Put Eigen in MPL2 mode

jwnimmer-tri, 2016-11-08:
Eigen obeys an `EIGEN_MPL2_ONLY` define, which I suggest we set when using the superbuild, because it would be nice to have Drake-the-library offer a non-viral mode, and Eigen is a required dependency.  In other words, Drake code should only depend on the MPL'd bits of Eigen.

Note that Drake-the-library should not set `EIGEN_MPL2_ONLY` directly (e.g., don't put it in a header file), to allow for embedding of Drake-the-library with a different Eigen selection in others' superbuilds.  Just that _our_ superbuild (and CI) should enforce the no-LGPL constraint.

Relates to #4045.

sherm1, 2016-11-08:
This seems like a good idea -- Eigen says all we lose is "SimplicialCholesky, AMD ordering and constrained_cg". Does anyone care about those?

david-german-tri, 2016-11-08:
Makes sense to me.

david-german-tri, 2016-11-08:
@mwoehlke-kitware Would you mind figuring out how to spell #4078 in CMake?

jamiesnape, 2016-11-08:
`add_definitions(-DEIGEN_MPL2_ONLY)`

jwnimmer-tri, 2016-11-08:
Thanks, that has unblocked me.

PR #4084 (2016-11-09): Use EIGEN_MPL2_ONLY
Closes #4065.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4084)
closed 2016-11-09 by jwnimmer-tri