RobotLocomotion/drake #4074: Use Embedded Ruby to Remove Duplication in the various Prius SDF Files

liangfok, 2016-11-08:
In [drake/automotive/models/prius](https://github.com/RobotLocomotion/drake/tree/master/drake/automotive/models/prius) there are multiple Prius SDF files that contain duplicate information. Let's remove the duplication using embedded ruby as described here:

See: http://answers.gazebosim.org/question/7213/parameterizing-elements-of-an-sdf-model/

david-german-tri, 2016-11-22:
> ...Embedded Ruby...

?! :running_man: :fire: :no_entry_sign: :scream: 

Joking aside, code generation for SDF is an OK idea, and if ERuby is an official recommendation from OSRF, I agree we should follow it.  Please aim for a pattern similar to `lcm_vector_gen`, though, where the generated outputs go in a `gen/` directory, and the generator tool is in-tree.  That makes it easy to migrate to a Bazel genrule in the future.

jwnimmer-tri, 2016-12-23:
The linked examples show how to de-duplicate information within an SDF using ERuby (i.e., support named variables within a file, so that literal data isn't monotonously / brittlely repeated).  I don't (yet?) see an example of ERuby that allows for an ensemble of SDFs to be created from a single description file plus parameters and/or config knobs.

How important is it to have the ensemble of separate SDFs on disk (prius.sdf, prius_with_lidar.sdf, prius_with_two_lidars.sdf, prius_with_lidar_that_jake_installed_tilted.sdf, prius_with_missing_fender_plus_radar.sdf, prius_with_side_mirror_smashed_off.sdf)?  Would it be acceptable instead to just have one prius.sdf master with just the basic car, and then at runtime add the sensors, based on program options, a config file, etc?

liangfok, 2016-12-23:
I think it's very desirable to consolidate all of the disparate Prius SDF models in `master` into something that's automatically generated. I'm not yet sure what the best approach for this will be. Programmatically modifying a model instance within a `RigidBodyTree` after a generic model specification is loaded also seems to suffer maintainability problems since we won't be able to determine what is being models by simply inspecting a text file (In URDF-land, we can use `xacro` to export it's auto-generated model into a text file, which we could then inspect).

Perhaps the best option will is to allow programmatic modification of a model instance within a `RigidBodyTree`, but then provide a `RigidBodyTree::ExportToSdf()` method that saves the current model instances within the `RigidBodyTree` into an SDF file that we could then inspect.

jwnimmer-tri, 2016-12-23:
What if for now we just delete all Prius SDFs and URDFs from the repo that are not `prius_with_lidar.sdf`?  They don't appear to be used for anything.

liangfok, 2016-12-23:
Sure, SGTM.

liangfok, 2016-12-23:
Actually, `drake-distro/drake/multibody/rigid_body_system1/test/CMakeLists.txt` uses "prius.sdf" and "prius.urdf":

https://github.com/RobotLocomotion/drake/blob/f6ac89600b28234b47047e3a8757a0c9473d2790/drake/multibody/rigid_body_system1/test/CMakeLists.txt#L24

We will need to make a `prius_with_lidar.urdf` if we want to remove `prius.sdf` and `prius.urdf`. Or we can delete the unit test. Or port it to System 2.0.

jwnimmer-tri, 2016-12-23:
If we have two models of the same thing, a regression test to keep them sync'd makes sense.  If we only need one copy of a model, probably we shouldn't a second copy in order to test that two copies stay in sync.

liangfok, 2016-12-23:
I believe the original goal of that unit test was to check whether the SDF parser, which was newer and more buggy at the time, produced the same model instance as the URDF parser, which was older and more-mature at the time. Maybe we can conclude that this motivation is no longer present and remove this unit test.

jwnimmer-tri, 2016-12-24:
Well, if we want to sanity-check the two parser implementations against each other, that's fine, but doing so using the `automotive` models is a drag on `automotive` development (it makes it difficult to edit the models).

We should remove the unwanted models, and if necessary file an issue that the parsers need better regression tests, using models that are curated for that purpose specifically.

jwnimmer-tri, 2016-12-27:
I took over this issue to understand it and assess whether we should close it.  I'm convinced the model file duplication is real issue and relatively trivially fixed, so this should stay open, and reassign over to @liangfok again as the responsible party again.

PR #9844 (2018-10-30): automotive: Remove unused models
Closes #4074.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9844)
closed 2018-10-30 by jwnimmer-tri