RobotLocomotion/drake #4076: Rigid-Body Force Element needs port to System2.0

foehnph, 2016-11-08:
This issue is to discuss the design of a RigidBodyForceElement. One possibility is a Leafsystem which has as input the state of the RigidBodySystem x and the control input u and it should generate an output B(q)*u.

This leaf system can then be combined in the Diagram with the source RBS.

This leaf system should be generated from parsing the URDF/SDF with the logic sitting inside parser_urdf.h/parser_sdf-h

(tagging @naveenoid )

naveenoid, 2016-11-08:
Tagging : @liangfok and @david-german-tri 

The first implementation can be tested with the Quadrotor examples. I can work with @foehnph in resolving this once you guys have design feedback.

liangfok, 2016-11-08:
>One possibility is a Leafsystem which has as input the state of the RigidBodySystem x and the control input u and it should generate an output B(q)*u.

Instead of "RigidBodySytsem" I believe you mean "RigidBodyPlant".

In the above statement, what are variables `B` and `u`?

foehnph, 2016-11-08:
Thanks for the correction.
`u` is the input to the system and `B(q)` maps this input back into full general coordinates for the plant.
Therefore `u`,`q` could be the inputs to a LeafSystem which generate `B(q)*u` as an output.
This output can then be connected to the input of a RigidBodyPlant.

liangfok, 2016-11-08:
Just to be clear, is the input, `u`, a vector of force/torque values that specify the force/torque the `RigidBodyForceElement` is asserting on bodies within the `RigidBodyPlant`?

sherm1, 2016-11-08:
@amcastro-tri, do you any thoughts about how this should be done?

amcastro-tri, 2016-11-08:
I was talking about this over to lunch with @RussTedrake today. For the case of the quadrotor we would a `DragForceSystem` (probably inheriting from a `ForceSystem`?) that applies the appropriate external force on a given rigid body given its velocity (state) and orientation. Also I guess for the quadrotor you would need something like an `ConstantForceSystem`? for the propellers. 

For something like that I think the input to these force systems could directly be the kinematics of the particular rigid body the force is applied to. In that way `RigidBodyPlant` is responsible for computing this kinematics information and can use caching to do it efficiently. Also the output to the `ForceSystem` could directly be the applied wrench on the body. For that we would probably need to implement input ports on `RigidBodyPlant` for external wrenches. The computation of generalized forces is then done by `RigidBodyPlant` in one place and therefore computations are reused (imagine you could have several `ForceSystem`'s and it would be inefficient having each of those computing the generalized forces for the RBP). 
I am just brainstorming here, thoughts?

sherm1, 2016-11-08:
I like that idea, @amcastro-tri. The complete force interface to RBPlant should be (1) generalized forces applied to particular dofs (mobilities) and (2) a collection of spatial forces (torque, force) applied at specified points on specified rigid bodies. That way RBTree can do all the messy calculations, and can correctly calculate reaction forces that result from those spatial forces, which it can't do if they are converted first to generalized forces.

naveenoid, 2016-11-09:
I mostly agree with your plan @amcastro-tri but what about stuff like actuator dynamics etc? I guess we need to then build a custom "source" system which needs to obtain its parameters from urdf / sdf and converts actuation input (currents / voltages / direct positions?) into spatial forces mapping into the corresponding actuator ports. 

Any idea when you might work on this? Or can a part of this task be shared? What would be the best design for RBP? in construction, the URDF needs to be parsed to open sufficient input ports right?

amcastro-tri, 2016-11-09:
Not a plan yet, I was just openly brainstorming to get the feedback from everybody interested on this. 
I think we should start simple and what the things we need first. Correct me if I am wrong but it seems like the immediate need for the quadrotor example are force systems to model drag and thrust forces.

When you say actuator dynamics you mean for instance modelling a DC motor? in any case, I think we should tackle first the problems we have a direct application/need for, in this case constant forces and drag.

siyuanfeng-tri, 2016-11-14:
For actuator dynamics, I believe @tkoolen's plan is to compute all the messy actuator stuff outside of RBT, the only input for RBT will be torque. Since actuator dynamics will require additional states and so on, and before we get into coupled joints, it's reasonable to assume independence between joints. #4004 has a sketch of the setup in sys2.

liangfok, 2016-11-14:
+1 for keeping the messy actuator stuff outside of the RBT.

RussTedrake, 2016-11-15:
to clarify, i think the proposed input to rigidbodyplant is the generalized forces (torques) + wrenches that can be applied to particular frames.  The list of these inputs (e.g. which joints have actuators, which points on which bodies can take external wrenches) would be built up programmatically.

naveenoid, 2017-05-12:
@amcastro-tri What is the status of the proposal discussed here? Is it possible to apply a generalized force + torque to a select frame on a select model instance on a RBtree within a RBPlant? If not, could we have a quick f2f soon on what needs to land for that? 
A student here in Zurich is very interested in that.

amcastro-tri, 2017-05-12:
As we discussed f2f with you here in CAM during your visit, there has not been an immediate need for this to get resolved. My goal was to get all of this functionality ASAP in MBT. However, if there is an immediate need we can talk about how this should look like.

sherm1, 2018-11-09:
Won't be adding this to RBPlant -- MultibodyPlant yes.

closed 2018-11-09 by sherm1