RobotLocomotion/drake #4080: Generalize Name of lcmt_iiwa_status and lcmt_iiwa_command

liangfok, 2016-11-08:
The following two files define LCM messages that apply to any force/torque controlled robot:

* [drake/lcmtypes/lcmt_iiwa_command.lcm](https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_iiwa_command.lcm)
* [drake/lcmtypes/lcmt_iiwa_status.lcm](https://github.com/RobotLocomotion/drake/blob/master/drake/lcmtypes/lcmt_iiwa_status.lcm)

Thus, we should re-name them to not be KUKA iiwa-specific. How about the following?

* `lcmt_robot_command.lcm`
* `lcmt_robot_status.lcm`

We can always add more fields to the messages to support more robots in the future. (I'm thinking of velocity commands / status values).

liangfok, 2016-11-08:
@sammy-tri, just wondering what you think about this.

sammy-tri, 2016-11-08:
Huh...   I agree it's fairly generic, but the current messages have a (nearly) 1:1 correspondence with the KUKA API.  I suppose the question is if that's actually valuable.

I feel like the generic version already exists to an extent (and I don't think I knew about it when I created these messages) in https://github.com/mwoehlke-kitware/bot_core_lcmtypes/blob/master/lcmtypes/bot_core_robot_state_t.lcm (though I don't know if I want to replace the iiwa messages with that either).

So, uh...   I guess I have some thoughts about this, I'm just not sure how useful they are.

jwnimmer-tri, 2016-11-08:
FWIW, I suggest waiting to generalize until there are three identified users for the generic message.  (In software, you should have three points to form a line, because points are noisy.)

liangfok, 2016-11-08:
Looking at [bot_core_robot_state_t.lcm](https://github.com/mwoehlke-kitware/bot_core_lcmtypes/blob/master/lcmtypes/bot_core_robot_state_t.lcm), I think it's general except for the [force_torque](https://github.com/mwoehlke-kitware/bot_core_lcmtypes/blob/master/lcmtypes/bot_core_force_torque_t.lcm) field, which assumes the robot has two feet and two hands :smile: .

I suppose since each robot may have a different set of sensors, it'll be hard to generalize the robot status messages.

RussTedrake, 2016-11-18:
fwiw -- @tkoolen and I have talked about this, too and had some plans for a more general robot state and robot command.

tkoolen, 2016-11-21:
From an email I sent to Russ:

> I am still using robot_state_t, but I really dislike that message type:
> 
> * Serializing and deserializing between that message type and the format I want should be 5 lines at most, but is currently much harder (e.g. conversion between v and twist + non-floating joint velocities, for all three cases of 'floating joint type').
> * Its current constraints require me having to check if the RigidBodyTree is actually compatible with the message type first.
> * The 'twist' field contains no frame information; it took me some time to figure out that in oh-distro, it is assumed to be expressed in a world-aligned body frame, which is kind of nonstandard.
> In my opinion, we shouldn't be sending joint name strings over the network all the time; this is currently done to make it somewhat easier to interpret LCM logs (as in: user looks up index of joint name, then is able to interpret the other information in robot_state_t).
> 
> Although we often think about a rigid body plant's state as being just q and v, note that Atlas and Valkyrie control require joint efforts to be part of the message they receive (so as not to have to worry about data synchronization), so we'll have to think about how to handle that part of the robot's state.
> 
> robot_command_t isn't so terrible, but:
> * Its name is bad; perhaps it should be renamed to humanoid_qp_control_robot_command_t or something else more specific.
> * Same note about sending joint names in every message

sammy-tri, 2019-02-14:
Closing because I think the current messages (particularly status) are specifically tied to the iiwa's FRI API.  The Jaco arm was pretty similar to an iiwa, but the APIs were still different enough that I felt different low-level control messages were appropriate.

closed 2019-02-14 by sammy-tri