RobotLocomotion/drake #4090: Mysterious CI Error: FAILED: external/src/CAPD4-stamp/CAPD4-build

liangfok, 2016-11-09:
# The Error

```
09:46:28 FAILED: external/src/CAPD4-stamp/CAPD4-build 
09:46:28 cd /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal/external/src/CAPD4 && "/usr/local/Cellar/cmake/3.7.0-rc3/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal --output /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build --  /usr/local/Cellar/cmake/3.7.0-rc3/bin/cmake -P /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-Debug.cmake && "/usr/local/Cellar/cmake/3.7.0-rc3/bin/ctest" --launch --target-name TARGET_NAME --build-dir /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal --output /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build --  /usr/local/Cellar/cmake/3.7.0-rc3/bin/cmake -E touch /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build
09:46:28 CMake Error at /Users/elcapitan/workspace/mac-clang-ninja-continuous-debug/build/externals/dreal/external/src/CAPD4-stamp/CAPD4-build-Debug.cmake:16 (message):
09:46:28   Command failed: 2
09:46:28 
09:46:28    'make'
```

# Example Logs
* https://drake-jenkins.csail.mit.edu/job/mac-clang-ninja-continuous-debug/838/console

soonho-tri, 2016-11-18:
I couldn't find a way to reproduce the problem. Apparently, re-triggering build worked. Let me close this for now. 

FYI, the use of CAPD in dReal is disabled for drake due to some possible auto-hell issues. @mwoehlke-kitware and I are working on a fix.

closed 2016-11-18 by soonho-tri

mwoehlke-kitware, 2016-11-18:
This... *might* actually be caused by an XCode update, where it wants to ask the user to accept the license. From the log, it appears that `make` failed but produced no output, which should not happen under normal circumstances, but ISTR that you do actually see similar symptoms when XCode is balking as described. At any rate, it doesn't seem reproducible; I don't believe it has occurred again in CI since the original report.

> FYI, the use of CAPD in dReal is disabled for drake

Pedantic: #4141 (not merged as of writing) will disable CAPD.

> ...due to some possible auto-hell issues

FTR, the problems I ran into were trying to build dReal by itself in an environment not set up specifically for drake, so the problems I was having may be related to the version of autotools I was using.

soonho-tri, 2016-11-18:
@mwoehlke-kitware, thanks for the clarification!

jamiesnape, 2016-11-18:
> This... might actually be caused by an XCode update, where it wants to ask the user to accept the license.

No updates recently and auto-updating is disabled.